openai/roboschool #99: install error

haoyuhaoyuhaoyu, 2017-12-03:
when come with the apartment ''pip install $ROBOSCHOOL''we met with a problem that  ''fatal error: 'QtWidgets/QOpenGLWidget' file not found''

yuchen93, 2017-12-14:
Same issue.

dcalacci, 2017-12-14:
I've found you need Qt version 5.4 or above -- check which you have installed!

haoyuhaoyuhaoyu, 2018-01-03:
my Qt version is 5.6.2,still not work

cshang412, 2018-01-11:
i am a beginner to RL ,today when i install roboschool i met some erro :+1: 
ubuntu16.04 
when i follow the methods in tutorial ,when i input the last order ：
 pip3 install -e $ROBOSCHOOL_PATH
it shows ：
'/path/to/roboschool' is not installable. File 'setup.py' not found. 
i tryed to change the location(i established  a new folder in Document ,and i colned the git there and changed the PATH to  ROBOSCHOOL_PATH=~Document/Git/roboschool （～Document/Git is where i colned the depo）

what confused me is i can "import gym " in python but cant import roboschool??
and i really creat a folder in that(~Document/Git/roboschool) location , but it shows that it doesn't has setup.py file ,is there anyone can help me sove the problem? thanks 
 )  it  still doesn't work ,i dont know why the setup.py is not incklued in the

YimiWangBJ, 2018-01-12:
@cshang412  I met the same error.File 'setup.py' not found when pip 3 install -e $ROBOSCHOOL_PATH.
my setuo.py is in $ROBOSCHOOL/bullet3,I change the path to /bullet3. Then successfully installed pybullet.But also can't import roboschool.I guess the Interpreter don't include the roboschool.

khoa-ngo, 2018-01-24:
have you tried:
brew install pkg-config

khoa-ngo, 2018-01-31:
Hi cshang412 and YimiWangBJ,

I think it's best to completely remove roboschool and bullet3 and start over.

for your case, cshang412, you would need to define ROBOSCHOOL_PATH at the start of the installation, this path is used later to install bullet.
```
export ROBOSCHOOL_PATH=~Document/Git/roboschool 
cd $ROBOSCHOOL_PATH
ls
```
you should see: 
```
LICENSE.md		agent_zoo		roboschool.egg-info
README.md		roboschool		setup.py
```
If otherwise, I suspect that your ROBOSCHOOL_PATH may be incorrect. First, navigate to your cloned roboschool folder in terminal, then:
```
pwd
```
output should be something like: /Users/anon/Documents/GitHub/roboschool

Now export your ROBOSCHOOL_PATH to that path and perform the [installation](https://github.com/openai/roboschool) again.