openai/roboschool #100: How to use the gym's Monitor wrapper on roboschool environments

syllogismos, 2017-12-05:
I'm not able to reset the wrapped roboschool environment more than once. How do I make roboschool work with Monitor wrapper. In the demos, they used video recorder seperately, I want to be able to use the monitor wrapper.

```
$ xvfb-run -s "-screen 0 1400x900x24" ipython
Python 3.6.0 | packaged by conda-forge | (default, Feb  9 2017, 14:36:55)
Type 'copyright', 'credits' or 'license' for more information
IPython 6.2.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import roboschool, gym

In [2]: env = gym.make('RoboschoolHumanoid-v1')

In [3]: envm = gym.wrappers.Monitor(env, 'test_monit', video_callable=lambda x: True, force=True)

In [4]: o = envm.reset()

In [5]: o = envm.reset()
render-glwidget.cpp:56 ERROR: 0x502
python: render-glwidget.cpp:56: void Household::Camera::camera_render(const boost::shared_ptr<SimpleRender::Context>&, bool, bool, bool): Assertion `e == GL_NO_ERROR' failed.
Aborted (core dumped)
```

```
$ xvfb-run -s "-screen 0 1400x900x24" ipython
Python 3.6.0 | packaged by conda-forge | (default, Feb  9 2017, 14:36:55)
Type 'copyright', 'credits' or 'license' for more information
IPython 6.2.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import roboschool, gym

In [2]: env = gym.make('RoboschoolHumanoid-v1')

In [3]: envm = gym.wrappers.Monitor(env, 'test_monit', video_callable=lambda x: True, force=True)

In [4]: o = envm.reset()

In [5]: done = False

In [6]: while not done:
   ...:     _, _, done, _ = envm.step(envm.action_space.sample())
   ...:

In [7]: o = envm.reset()
render-glwidget.cpp:56 ERROR: 0x502
python: render-glwidget.cpp:56: void Household::Camera::camera_render(const boost::shared_ptr<SimpleRender::Context>&, bool, bool, bool): Assertion `e == GL_NO_ERROR' failed.
Aborted (core dumped)
```

ethanluoyc, 2017-12-21:
I can confirm that I am running into similar problems. However, I believe that the reason this is failing is because Monitor is calling `render` with argument 'rgb_array', which for some reason causes this to fail.

trane293, 2018-04-03:
Did you solve this? I'm having the same issue!