openai/roboschool #103: Running on Mac Problem / render-simple.cpp:540 ERROR: 0x506

hiwonjoon, 2017-12-19:
Hello,

When I run the test `python agent_zoo/demo_race2.py` after successfully install roboschool, I got the message:

```
render-simple.cpp:540 ERROR: 0x506
Assertion failed: (e == GL_NO_ERROR), function paint, file render-simple.cpp, line 540.
[1]    86412 abort      python agent_zoo/demo_race2.py
```

I am currently on newest MacBook Pro 15 with High Sierra. I installed the required libraries using homebrew. I was trying to locate the error, but I couldn't.

Any ideas will be welcomed. Thanks!

Jeff-Shim, 2017-12-19:
This is happening to me as well with Mid-2014 MBPR 15-inch High Sierra with Geforce GT 750M. Seems like mac-oriented problem since everything's fine with my multiple Ubuntu machines.

Jeff-Shim, 2017-12-19:
After commenting out CHECK_GL_ERROR at render-simple.cpp line 540 and other multiple locations, the demos are working. While looking into detail, this might be an ugly workaround believing there are no errors to check 😈.

hiwonjoon, 2017-12-19:
Thanks, @Jeff-Shim! It seems very weird, but I confirmed that just commenting out assertion code for `GL_NO_ERROR` make the codes work.

I guess, the culprit is High Sierra since my friend who has the exact same model with Sierra does not have any problem.

khoa-ngo, 2018-01-31:
hi hiwonjoon,

I ran into the same issue with High Sierra. The issue was solved after downgrading to Sierra 10.12.6.

I performed the downgrade with [macOS Recovery](https://support.apple.com/en-us/HT201314)

wangshub, 2018-04-02:
@minostauros that works for me

wangshub, 2018-04-02:
- macOs High Sierra 10.13.3

jymagic, 2018-10-11:
hi guys, I am getting a very similar error, except that its at line 539

render-simple.cpp:539 ERROR: 0x506
Assertion failed: (e == GL_NO_ERROR), function paint, file render-simple.cpp, line 539.

What I don't understand is that i commented out all of the GL_NO_ERROR and CHECK_GL_ERROR lines, I even deleted the render-simple.cpp! But the same error persists

[image]

jymagic, 2018-10-11:
@minostauros @hiwonjoon also worked for me. In addition to commetting out CHECK_GL_ERROR in render-simple.cpp, I also commented out all of the GL_NO_ERROR in render-glwidget.cpp, and it worked like a charm. Thanks guys!