openai/roboschool #110: Qt5 error during pip install -e $ROBOSCHOOL_PATH

FitMachineLearning, 2018-01-21:
I have followed README instructions multiple times but I am not able to find solution for this. On the final step of the install during the "sudo pip install -e $ROBOSCHOLL_PATH" command, I get the following error.

I am on python 3.6 with Anaconda on Mac High Sierra.

" [...]
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
    Package Qt5Widgets was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5Widgets.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5Widgets' found
    Package Qt5OpenGL was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5OpenGL.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5OpenGL' found
    Package python-3.5 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.5.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.5' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
    Package Qt5Widgets was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5Widgets.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5Widgets' found
    Package Qt5OpenGL was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5OpenGL.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5OpenGL' found
    Package python-3.5 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.5.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.5' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
    Package Qt5Widgets was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5Widgets.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5Widgets' found
    Package Qt5OpenGL was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5OpenGL.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5OpenGL' found
    Package python-3.5 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.5.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.5' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-glwidget.cpp  -o.build-release/render-glwidget.o -MMD -MF .build-release/render-glwidget.o.dep
    Package Qt5Widgets was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5Widgets.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5Widgets' found
    Package Qt5OpenGL was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5OpenGL.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5OpenGL' found
    Package python-3.5 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.5.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.5' found
    In file included from physics-bullet.cpp:1:
    In file included from ./render-glwidget.h:1:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    make: *** [.build-release/physics-bullet.o] Error 1
    make: *** Waiting for unfinished jobs....
    In file included from assets-mesh.cpp:1:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    In file included from random-world-tools.cpp:3:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    make: *** [.build-release/random-world-tools.o] Error 1
    In file included from render-glwidget.cpp:2:
    In file included from ./render-glwidget.h:1:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    1 error generated.
    make: *** [.build-release/render-glwidget.o] Error 1
    make: *** [.build-release/assets-mesh.o] Error 1
    cd /Users/eljed/packages/roboschool3/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /Users/eljed/packages/roboschool3/roboschool/
Eljeds-iMac:build eljed$

TheCrazyT, 2018-01-21:
Check your PKG_CONFIG_PATH environment variable.
Inside that path there must be a Qt5Widgets.pc. (basicalls thats what the error message tells you)
It probably isn't, so:
- you maybe for got to install Qt5
  do:
```
   sudo apt-get install qt5-default
```
- forgot to set the right path:
   do:
```
    cd /
    find -name Qt5Widgets.pc
    export PKG_CONFIG_PATH=HERE_YOUR_PATH_TO_TO_FOUND_FILE:$PKG_CONFIG_PATH
```
for example if you find the file inside: /usr/lib/x86_64-linux-gnu/pkgconfig/ it would be:
 ```
  export PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig/:$PKG_CONFIG_PATH
 ```

FitMachineLearning, 2018-01-22:
Thanks @TheCrazyT 
`$ Brew install qt5 pkg-config`
yields
> Warning: qt 5.10.0_1 is already installed
> Warning: pkg-config 0.29.2 is already installed
```
$ cd $PKG_CONFIG_PATH 
$ ls
```
[...]
Qt5WebView.pc			ncursesw.pc
Qt5Widgets.pc			openssl.pc
Qt5Xml.pc			panel.pc
Qt5XmlPatterns.pc		panelw.pc
dbus-1.pc
[...]

So i think it is pointing to the right folder. Unless I missed something.

haoyuhaoyuhaoyu, 2018-01-24:
I have met the same problem ,have you solved the problem?

khoa-ngo, 2018-01-31:
Hi FitMachineLearning,

I recommend downgrading to Sierra since I never got roboschool to work with High Sierra. You would have to perform a clean install to downgrade, [tutorial here](https://www.imore.com/how-downgrade-macos)

You should also downgrade python to 3.5.2 within Anaconda
```
conda install python=3.5.2
```
Also make sure you've defined $PKG_CONFIG_PATH within the same session that where you're performing installation.

Have you tried installing with pip3?
```
pip3 install -e $ROBOSCHOOL_PATH
```

It would be useful to know your:
```
pip list
pip3 list
python -V
python3 -V
```

Phylliida, 2018-03-03:
The following solved it for me: (perhaps one of the paths here isn't necessary idk)
```
export PKG_CONFIG_PATH=/usr/local/opt/qt5/lib:/usr/local/opt/qt5/lib/QtWidgets.framework:/usr/local/opt/qt5/lib/pkgconfig

FitMachineLearning, 2018-03-13:
Thanks, @khoa-ngo .

Downgrading to Sierra isn't an option for me. So I will try these instructions on High Sierra.

FitMachineLearning, 2018-03-13:
@Phylliida thanks for the help.
Will try that when I am back at my coding station.

I must admit, this has been a very frustrating experience.

FitMachineLearning, 2018-03-19:
Thanks a bunch

FitMachineLearning, 2018-03-19:
Still no joy.

I get the following...

`
(p35) bash-3.2$ cd /usr/local/opt/qt
(p35) bash-3.2$ pwd
/usr/local/opt/qt
(p35) bash-3.2$ export PKG_CONFIG_PATH=/usr/local/opt/qt:/usr/local/opt/qt/lib/QtWidgets.framework/:/usr/local/opt/qt/lib/pkgconfig/
(p35) bash-3.2$ sudo pip3 install -e $ROBOSCHOOL_PATH
The directory '/Users/eljed/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/eljed/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Obtaining file:///Users/eljed/packages2/roboschool
    Complete output from command python setup.py egg_info:
    found resource files: 185
    running egg_info
    Mac, assumed python3, homebrew branch
    Link against python 3.5
    Slow hardware or software render (no shadows)
    rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so ../cpp_household_d.so  .build-release/physics-bullet.o  .build-release/assets-mesh.o  .build-release/random-world-tools.o  .build-release/render-glwidget.o  .build-release/render-hud.o  .build-release/render-simple.o  .build-release/render-simple-primitives.o  .build-release/test-tool-qt4.o  .build-release/python-binding.o  .build-debug/physics-bullet.o  .build-debug/assets-mesh.o  .build-debug/random-world-tools.o  .build-debug/render-glwidget.o  .build-debug/render-hud.o  .build-debug/render-simple.o  .build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o  .build-debug/python-binding.o .generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep .build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep .build-release/render-glwidget.o.dep .build-release/render-hud.o.dep .build-release/render-simple.o.dep .build-release/render-simple-primitives.o.dep .build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep .build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep .build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep .build-debug/render-hud.o.dep .build-debug/render-simple.o.dep .build-debug/render-simple-primitives.o.dep .build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep
    rm -rf .generated
    rm -rf .build-debug
    rm -rf .build-release
    Mac, assumed python3, homebrew branch
    Link against python 3.5
    Slow hardware or software render (no shadows)
    mkdir -p .generated
    mkdir -p .build-release
    mkdir -p .build-debug
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
    Package Qt5Widgets was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5Widgets.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5Widgets' found
    Package Qt5OpenGL was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5OpenGL.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5OpenGL' found
    Package python-3.5 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.5.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.5' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
    Package Qt5Widgets was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5Widgets.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5Widgets' found
    Package Qt5OpenGL was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5OpenGL.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5OpenGL' found
    Package python-3.5 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.5.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.5' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
    Package Qt5Widgets was not found in the pkg-config search path.
[... cut ...]
    cd /Users/eljed/packages2/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /Users/eljed/packages2/roboschool/

`

Phylliida, 2018-03-20:
Try running your commands with

sudo -H

instead of sudo

FitMachineLearning, 2018-03-21:
Thanks Phyllida, Tried sudo -H,
This is what I get.

---

(p35) bash-3.2$ sudo -H pip3 install -e $ROBOSCHOOL_PATH

Password:

Obtaining file:///Users/eljed/packages2/roboschool

    Complete output from command python setup.py egg_info:

    found resource files: 185

    running egg_info

    Mac, assumed python3, homebrew branch

    Link against python 3.5

    Slow hardware or software render (no shadows)

    rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so
../cpp_household_d.so  .build-release/physics-bullet.o
.build-release/assets-mesh.o  .build-release/random-world-tools.o
.build-release/render-glwidget.o  .build-release/render-hud.o
.build-release/render-simple.o  .build-release/render-simple-primitives.o
.build-release/test-tool-qt4.o  .build-release/python-binding.o
.build-debug/physics-bullet.o  .build-debug/assets-mesh.o
.build-debug/random-world-tools.o  .build-debug/render-glwidget.o
.build-debug/render-hud.o  .build-debug/render-simple.o
.build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o
.build-debug/python-binding.o
.generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep
.build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep
.build-release/render-glwidget.o.dep .build-release/render-hud.o.dep
.build-release/render-simple.o.dep
.build-release/render-simple-primitives.o.dep
.build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep
.build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep
.build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep
.build-debug/render-hud.o.dep .build-debug/render-simple.o.dep
.build-debug/render-simple-primitives.o.dep
.build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep

    rm -rf .generated

    rm -rf .build-debug

    rm -rf .build-release

    Mac, assumed python3, homebrew branch

    Link against python 3.5

    Slow hardware or software render (no shadows)

    mkdir -p .generated

    mkdir -p .build-release

    mkdir -p .build-debug

    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function
-Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3
-march=native -I/System/Library/Frameworks/OpenGL.framework/Headers
`pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5`
-Ibullet_local_install/include -Ibullet_local_install/include/bullet
-I/usr/local/include/bullet -c physics-bullet.cpp
[... 181 lines cut ...]

    cd /Users/eljed/packages2/roboschool/roboschool/cpp-household && make
clean && make -j4 dirs  ../cpp_household.so

    C++ dependencies for this project are:

    bullet

    tinyxml

    boost_python

    assimp

    Qt5

    If you see compilation error FIRST THING TO CHECK if pkg-config call
was successful.

    Install dependencies that pkg-config cannot find.

    ----------------------------------------

Command "python setup.py egg_info" failed with error code 1 in
/Users/eljed/packages2/roboschool/

FitMachineLearning, 2018-04-10:
Ok Last try at this.

Suspecting environment variable issues i did the following.

`export PKG_CONFIG_PATH=$PKG_CONFIG_PATH=/Users/eljed/anaconda3/pkgs/qt-5.6.2-h9975529_14/lib/pkgconfig:/Users/eljed/anaconda3/pkgs/qt-5.6.2-h9975529_14/lib:/usr/local/opt/qt/lib:/usr/local/opt/qt/lib/QtWidgets.framework:/Users/eljed/anaconda3/lib/cmake/Qt5Widgets:/Users/eljed/anaconda3/pkgs/qt-5.6.2-h9975529_14/lib/pkgconfig:/Users/eljed/anaconda3/pkgs/qt-5.9.4-h14e15a9_0/lib:/Users/eljed/anaconda3/pkgs/qt-5.9.4-h14e15a9_0/lib/pkgconfig:/Users/eljed/anaconda3/envs/py35rs/lib/pkgconfig`

Unfortunately this still gives me the following error:
Your help would be greatly appreciated

(py35rs) Eljeds-iMac:lib eljed$ sudo -H pip3 install -e $ROBOSCHOOL_PATH
Password:
Obtaining file:///Users/eljed/packages3/roboschool/roboschool
    Complete output from command python setup.py egg_info:
    found resource files: 185
    running egg_info
    Mac, assumed python3, homebrew branch
    Link against python 3.5
    Slow hardware or software render (no shadows)
    rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so ../cpp_household_d.so  .build-release/physics-bullet.o  .build-release/assets-mesh.o  .build-release/random-world-tools.o  .build-release/render-glwidget.o  .build-release/render-hud.o  .build-release/render-simple.o  .build-release/render-simple-primitives.o  .build-release/test-tool-qt4.o  .build-release/python-binding.o  .build-debug/physics-bullet.o  .build-debug/assets-mesh.o  .build-debug/random-world-tools.o  .build-debug/render-glwidget.o  .build-debug/render-hud.o  .build-debug/render-simple.o  .build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o  .build-debug/python-binding.o .generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep .build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep .build-release/render-glwidget.o.dep .build-release/render-hud.o.dep .build-release/render-simple.o.dep .build-release/render-simple-primitives.o.dep .build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep .build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep .build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep .build-debug/render-hud.o.dep .build-debug/render-simple.o.dep .build-debug/render-simple-primitives.o.dep .build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep
    rm -rf .generated
    rm -rf .build-debug
    rm -rf .build-release
    Mac, assumed python3, homebrew branch
    Link against python 3.5
    Slow hardware or software render (no shadows)
    mkdir -p .generated
    mkdir -p .build-release
    mkdir -p .build-debug
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
    Package Qt5Widgets was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5Widgets.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5Widgets' found
    Package Qt5OpenGL was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5OpenGL.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5OpenGL' found
    Package python-3.5 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.5.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.5' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
    Package Qt5Widgets was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5Widgets.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5Widgets' found
    Package Qt5OpenGL was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5OpenGL.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'Qt5OpenGL' found
    Package python-3.5 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.5.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.5' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
    Package Qt5Widgets was not found in the pkg-config search path.
    Perhaps you should add the directory containing `Qt5Widgets.pc'
[... cut ...]
    make: *** [.build-release/random-world-tools.o] Error 1
    make: *** [.build-release/render-glwidget.o] Error 1
    cd /Users/eljed/packages3/roboschool/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.

Phylliida, 2018-04-10:
You have the formatting wrong:

‘’’
export PKG_CONFIG_PATH=$PKG_CONFIG_PATH=/Users/...
‘’’

Should be 

‘’’
export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/Users/...
‘’’

(Don’t have the second equals sign)

You might have hecked up that environment variable by putting an = in it, try starting a new terminal and then setting roboschool path and such again

FitMachineLearning, 2018-04-10:
Thanks Phyllida,
This has now been corrected but hasn't fixed the issue.

I have documented my full install process here
https://youtu.be/Tc9RAb9x5Qo

Hopefully you might be able to spot something else that leads me to this error.

if I am able to fix this error, I will create a video documenting the full correct install process to benefit all other researchers.

wsgdrfz, 2018-04-15:
Same problem..

FitMachineLearning, 2018-04-15:
OK at this point I am ready to pay someone some money to tell me what is still wrong here.

I have followed the instructions and recommendations of many here to avail

1 - I have reinstalled my mac
2 - I have installed a fresh copy of anaconda with 3.5.2
3 - I have set the PKG_CONFIG_PATH to where my Qt5Widget.pc file is (You can see that from the screen capture below).

https://i.imgur.com/jxDEdr6.png
https://i.imgur.com/Oenc1tu.png

I still get the same error.

Am I setting the wrong env variable?
Is there something I am missing.

I would really appreciate your help on this one.

olegklimov, 2018-04-15:
Hi @FitMachineLearning . It is usually not that hard to debug. What does `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` tell you when you run it in command line? Write a small "hello world" using Qt5. Will it compile if you put there the same flags as pkg-config tells you? You will probably see the same error. Follow the paths that it uses, and you'll find the problem.

FitMachineLearning, 2018-04-15:
Hi @olegklimov ,
Thanks for the attention to this issue. Many of us aren't that skilled at navigating MacOS issues.

When I run the above command this is what I get.

https://i.imgur.com/8ysOnQm.png

Not sure how to write a "hello world for Qt5. Doing you mean like this https://wiki.qt.io/Qt_for_Beginners

olegklimov, 2018-04-15:
Right, so it points to Qt inside Anaconda. If you want to do that, it's fine, but you've got to know what you are doing. This project is known to work with Qt as installed by homebrew.

FitMachineLearning, 2018-04-15:
OK,

I will restart the install with the mac-os homebrew install instruction and
report back in a bit.

FitMachineLearning, 2018-04-15:
I do get the exact same error.

See imaged ko
https://i.imgur.com/2W6vD7T.png

olegklimov, 2018-04-15:
Fix 'No package Python-3.5'.

pkg-config fails, doesn't produce any flags.

FitMachineLearning, 2018-04-15:
Excuse my ignorance here, 
But is this an option that I have to add to one of the install commands? If so where?

Should I be doing something like this ?

`pip install -e $ROBOSCHOOL_PATH No package Python-3.5`

FitMachineLearning, 2018-04-15:
My apologies Oleg, I just noticed that in my output it says

> Package python-3.5 was not found in the pkg-config search path.
>  Perhaps you should add the directory containing `python-3.5.pc'

I will find it and add it to the PKG_CONFIG_PATH and report back

FitMachineLearning, 2018-04-15:
OK, Progress!!!
I now get the following error.

https://i.imgur.com/1d2s4S3.png
https://i.imgur.com/Pzlm0R7.png

FitMachineLearning, 2018-04-16:
`brew reinstall boost-python --without-python --with-python3 --build-from-source`
running the above again didn't fix it.

https://i.imgur.com/BKm1PFV.png
https://i.imgur.com/SS0lf8e.png

FitMachineLearning, 2018-04-16:
Finally was able to install.

Following the recommendations found here [here](https://github.com/openai/roboschool/issues/118) I had to run 
`brew reinstall boost-python3 --with-python  --build-from-source`

I was finally able to run `pip install -e $ROBOSCHOOL_PATH`

I will report back if once I have been able to ran an example.

FitMachineLearning, 2018-04-16:
Thanks a million @olegklimov .
I was able to install.
I am running into a 1)segmentation fault 11 and 2): Library not loaded: @rpath/libQt5OpenGL.5.dylib
  Referenced from: /Users/eljedi2/package/roboschool6/gym/roboschool/roboschool/cpp_household.so

The issue is discussed [here](https://github.com/openai/roboschool/issues/31) so I will be closing this issue in a few days.

If there is anyway I can help, even if it is for financial contribution/donation, I would love to.

FitMachineLearning, 2018-04-16:
After full reinstall using Anaconda with a 3.5.2 environment using [these instructions](http://cslab.org/blog/roboschool), I finally got the demo working by setting the right environment variables

> export ROBOSCHOOL_PATH=/path/to/roboschool
> export PKG_CONFIG_PATH="/path/to/conda-environment/lib/pkgconfig"
> export LD_LIBRARY_PATH="/path/to/conda-environment/lib:$LD_LIBRARY_PATH"
> export LIBRARY_PATH="/path/to/conda-environment/lib:$LIBRARY_PATH"
> export C_INCLUDE_PATH="/path/to/conda-environment/include:/path/to/conda-environment/include/python3.6m"
> export CPLUS_INCLUDE_PATH="/path/to/conda-environment/include:/path/to/conda-environment/include/python3.6m"

In my case I didn't get have to use the python3.6m folders as I was doing a 3.5.2 install.

Also make sure you do the following.
> First, remove boost-python package, as boost package is a superset of that.
> Then, change [this](https://github.com/openai/roboschool/blob/master/roboschool/cpp-household/Makefile#L59) line to BOOST_PYTHON = -lboost_python3

You may need to recompile roboschool with `pip install -e $ROBOSCHOOL_PATH`

I got the instructions from [here](https://github.com/openai/roboschool/issues/122) 
thanks @nrontsis