openai/roboschool #111: Directory '/usr/local/roboschool' is not installable. File 'setup.py' not found.

lightning20, 2018-01-22:
The path ROBOSCHOOL_PATH has been set to '/usr/local/roboschool'. While using sudo also not working.

Any suggestion, thanks.

lightning20, 2018-01-22:
Tried pip3 install -e $ROBOSCHOOL_PATH again by changing the directory to /usr/local/roboschool/bullet3, where has a setup.py. 
It is successfully installed, but cannot find and run the humanoid example.

Any suggestion, thanks.

lightning20, 2018-01-22:
As the install is using pip3 for python3, but the local engaged environment is python 2.7.1, will there be any issues, thanks.

lightning20, 2018-01-22:
Actually, I am not very sure when download and unzip from zip. Should I copy all the files from the zip folder /home/d/Downloads/roboschool-master to the ROBOSCHOOL_PATH /usr/local/roboschool. Really appreciated for helping.

khoa-ngo, 2018-01-31:
Hey lightning20,

What is your current OS? I recommend upgrading your python and python3 to 3.5.2. You should use [Anaconda](https://anaconda.org/anaconda/anaconda-navigator) to avoid messing with the local system python.

It would also be a good idea to completely remove your roboschool and bullet3 folders and start over. This time, it would be better to use git:
```
cd
mkdir GitHub
cd GitHub
git clone https://github.com/openai/roboschool
cd roboschool
pwd
```
the commands did the following:
1. Navigate to your user directory
2. Make a new folder named "GitHub"
3. Navigate to new folder GitHub
4. Clone a copy of roboschool folder in your current folder GitHub
5. Navigate to roboschool folder
6. Output path for current folder

the output of "pwd" should be linked to ROBOSCHOOL_PATH. For me, it's:
```
export ROBOSCHOOL_PATH=/Users/khoa/Documents/GitHub/roboschool
```
Note: You should perform the installation within only one session (one terminal window) because the Bash export command may not translate between sessions.

Now you're good to proceed with the [installation](https://github.com/openai/roboschool).

Once you're done the following should not give you any issues:
```
python3
import roboschool
quit()
```

wych1005, 2018-12-29:
In some cases, it may be necessary to export the PKG_CONFIG_PATH variable with the directory name python-3.6.pc, for example:

`export PKG_CONFIG_PATH=/home/wych/anaconda/envs/robo/lib/pkgconfig/`