openai/roboschool #112: Import roboschool failure

swiftwing, 2018-01-22:
I have successfully installed roboschool on Ubuntu 16.04 under Anaconda python 3.5.

When I open a python terminal and import roboschool, there's no problem:
[image]
But when I try to run `python $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py`, I have this error:
[image]

I don't know what's happening here. Any help would be grateful.

khoa-ngo, 2018-01-24:
have you tried:
conda install libgcc

haoyuhaoyuhaoyu, 2018-01-24:
the same error

swiftwing, 2018-01-25:
I tried `conda install libgcc` but that does not help. The same things happens.

khoa-ngo, 2018-01-31:
Hi swiftwing,

Would you mind copying and pasting the output of:

```
pip list
pip3 list
python -V
python3 -V
```

I also recommend downgrading python to 3.5.2
```
conda install python=3.5.2
```

I suspect the following might work for you:
```
python3 $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
```

swiftwing, 2018-01-31:
# pip list:

actionlib (1.11.11)
alabaster (0.7.10)
anaconda-client (1.6.5)
anaconda-navigator (1.6.9)
anaconda-project (0.8.0)
angles (1.9.11)
asn1crypto (0.22.0)
astroid (1.5.3)
astropy (2.0.2)
atari-py (0.1.1)
Babel (2.5.0)
backports.shutil-get-terminal-size (1.0.0)
base-local-planner (1.14.2)
beautifulsoup4 (4.6.0)
bitarray (0.8.1)
bkcharts (0.2)
blaze (0.11.3)
bleach (1.5.0)
bokeh (0.12.10)
bondpy (1.8.1)
boto (2.48.0)
Bottleneck (1.2.1)
camera-calibration (1.12.22)
camera-calibration-parsers (1.11.13)
catkin (0.7.8)
catkin-pkg (0.4.1)
certifi (2017.11.5)
cffi (1.10.0)
chainer (3.3.0)
chardet (3.0.4)
click (6.7)
cloudpickle (0.4.0)
clyent (1.2.2)
colorama (0.3.9)
contextlib2 (0.5.5)
controller-manager-msgs (0.13.0)
cryptography (2.0.3)
cv-bridge (1.12.7)
cycler (0.10.0)
Cython (0.26.1)
cytoolz (0.8.2)
dask (0.15.3)
datashape (0.5.4)
decorator (4.1.2)
diagnostic-analysis (1.9.2)
diagnostic-common-diagnostics (1.9.2)
diagnostic-updater (1.9.2)
distributed (1.19.1)
docutils (0.14)
dynamic-reconfigure (1.5.48)
entrypoints (0.2.3)
enum34 (1.1.6)
et-xmlfile (1.0.1)
fastcache (1.0.2)
filelock (2.0.12)
Flask (0.12.2)
Flask-Cors (3.0.3)
future (0.16.0)
gazebo-plugins (2.5.14)
gazebo-ros (2.5.14)
gencpp (0.5.5)
geneus (2.2.6)
genlisp (0.4.16)
genmsg (0.5.9)
gennodejs (2.0.1)
genpy (0.6.7)
geodesy (0.5.2)
gevent (1.2.2)
[... 411 lines cut ...]
unity-scope-firefoxbookmarks (0.1)
unity-scope-gdrive (0.7)
unity-scope-manpages (0.1)
unity-scope-openclipart (0.1)
unity-scope-texdoc (0.1)
unity-scope-tomboy (0.1)
unity-scope-virtualbox (0.1)
unity-scope-yelp (0.1)
unity-scope-zotero (0.1)
urdfdom-py (0.3.3)
urllib3 (1.22)
usb-creator (0.3.0)
vcstool (0.1.32)
wheel (0.29.0)
xacro (1.11.2)
xdiagnose (3.8.4.1)
xkit (0.0.0)
XlsxWriter (0.7.3)

 
# python -V:
Python 3.5.4 :: Anaconda custom (64-bit)

# python3 -V:
Python 3.5.4 :: Anaconda custom (64-bit)

khoa-ngo, 2018-01-31:
Can you try redefining your ROBOSCHOOL_PATH again and do:
```
pip install -e $ROBOSCHOOL_PATH
```

swiftwing, 2018-02-01:
I just did it. Although it says successfully installed, when I run `pip list`, 'roboschool' is still not there. Also, when I run `python3 $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py `, it still says:
ImportError: No module named 'roboschool'.

khoa-ngo, 2018-02-01:
That's strange.. were you inside Anaconda environment during installation and running pip list?

swiftwing, 2018-02-01:
Yes, I was inside Anaconda environment.

khoa-ngo, 2018-02-01:
Maybe this could help you. I've just put together a Virtualbox image of Ubuntu 16.04 with working roboschool and gym, tested with Virtualbox Version 5.2.6 r120293: [Virtualbox download](https://www.virtualbox.org/wiki/Downloads), [Image download](https://www.dropbox.com/s/0fpmz01nu8mtqw6/roboschool-ubuntu16.04LTS.zip?dl=0), password: openai123.

This is what was done to set up the image:
1. Install [ubuntu-16.04.3-desktop-amd64](http://releases.ubuntu.com/16.04/) in new Virtualbox machine
2. Update package list and install newest package version:
```
sudo apt-get update
sudo apt-get upgrade
```
3. Install pip for python and python3 (Python versions 2.7.12 and 3.5.2 respectively)
```
sudo apt-get install curl
curl -O https://bootstrap.pypa.io/get-pip.py
sudo python get-pip.py
sudo python3 get-pip.py
```
4. Installing gym
```
sudo pip3 install gym
```
5. Clone roboschool
```
sudo apt-get install git
cd
mkdir GitHub
cd GitHub
git clone https://github.com/openai/roboschool
cd roboschool
pwd
```
6. Define ROBOSCHOOL_PATH
```
export ROBOSCHOOL_PATH=/home/roboschool/GitHub/roboschool
```
7. Install roboschool dependencies
```
sudo apt install cmake ffmpeg pkg-config qtbase5-dev libqt5opengl5-dev libassimp-dev libpython3.5-dev libboost-python-dev libtinyxml-dev
```
8. Install Bullet
```
cd
cd GitHub
git clone https://github.com/olegklimov/bullet3 -b roboschool_self_collision
mkdir bullet3/build
cd bullet3/build
cmake -DBUILD_SHARED_LIBS=ON -DUSE_DOUBLE_PRECISION=1 -DCMAKE_INSTALL_PREFIX:PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install -DBUILD_CPU_DEMOS=OFF -DBUILD_BULLET2_DEMOS=OFF -DBUILD_EXTRAS=OFF  -DBUILD_UNIT_TESTS=OFF -DBUILD_CLSOCKET=OFF -DBUILD_ENET=OFF -DBUILD_OPENGL3_DEMOS=OFF ..
make -j4
make install
cd ../..
```
9. Install roboschool
```
sudo pip3 install -e $ROBOSCHOOL_PATH
```
10. Testing
```
python3 $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
```

At the very least, I hope this can get you up and going with roboschool.

swiftwing, 2018-02-01:
Thank you very much.

jt-weller, 2018-02-26:
I see from the command 'pip list' that pluginlib (1.11.2) is listed. How can I download it?