openai/roboschool #118: brew reinstall boost-python with python 3?

jefio, 2018-02-11:
Using mac OS X 10.13.3, anaconda python 3. Following the installation instructions, `brew reinstall boost-python --without-python --with-python3 --build-from-source` succeeds but outputs the following warnings:

```
$ brew reinstall boost-python --without-python --with-python3 --build-from-source
==> Reinstalling boost-python
Warning: boost-python: this formula has no --with-python3 option so it will be ignored!
Warning: boost-python: this formula has no --without-python option so it will be ignored!
==> Downloading https://dl.bintray.com/boostorg/release/1.66.0/source/boost_1_66_0.tar.bz2
Already downloaded: /Users/jef/Library/Caches/Homebrew/boost-python-1.66.0.tar.bz2
==> ./bootstrap.sh --prefix=/usr/local/Cellar/boost-python/1.66.0_1 --libdir=/usr/local/Cellar/boost-python/1.66.0_1/lib --with-libraries=python --with-python=python
==> ./b2 --build-dir=build-python --stagedir=stage-python python=2.7 --prefix=/usr/local/Cellar/boost-python/1.66.0_1 --libdir=/usr/local/Cellar/boost-python/1.66.0_1/lib -d2 -j4
🍺  /usr/local/Cellar/boost-python/1.66.0_1: 457 files, 17.3MB, built in 4 minutes 43 seconds
```

After that, `pip install -e $ROBOSCHOOL_PATH` fails with the following error:

```
$ pip install -e $ROBOSCHOOL_PATH
[...]
    ld: library not found for -lboost_python3
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make: *** [../cpp_household.so] Error 1
    cd /Users/jef/work/code/source/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so

    C++ dependencies for this project are:

    bullet
    tinyxml
    boost_python
    assimp
    Qt5

    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /Users/jef/work/code/source/roboschool/
```

backpropper, 2018-02-12:
Same error! Even removing the prefix 3 does not help.

tmahlstrom, 2018-02-21:
Same problem here :/

jefio, 2018-02-22:
@cpacker Thank you! Your suggestion helped me run `pip3 install -e $ROBOSCHOOL_PATH` successfully. However I still ran into other problems afterwards. I ended up doing this:

- Note that I had to use Homebrew python instead of Anaconda python (because of this [problem](https://github.com/openai/roboschool/issues/31))
- In the installation instructions for Homebrew python, as suggested by @cpacker, I replaced `brew reinstall boost-python --without-python --with-python3 --build-from-source` by `brew install boost-python3`
- Before running `pip3 install -e $ROBOSCHOOL_PATH`, I had to edit `roboschool/cpp-household/render-simple.cpp` and `roboschool/cpp-household/render-glwidget.cpp` (see how and why [here](https://github.com/openai/roboschool/issues/103))

After that I was finally able to run the two examples from the README:

```
python $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
python $ROBOSCHOOL_PATH/agent_zoo/demo_race2.py
```

jinntrance, 2018-06-26:
After installing boost-python3 by `brew install boost-python3`, try the following code:
```bash
ln -s /usr/local/lib/libboost_python36.a /usr/local/lib/libboost_python3.a 
ln -s /usr/local/lib/libboost_python36.dylib /usr/local/lib/libboost_python3.dylib 
```