openai/roboschool #122: Install with conda commands only

nrontsis, 2018-02-26:
I managed to install using anaconda "only" (python 3.6) as following:

## Set paths
```
export ROBOSCHOOL_PATH=/path/to/roboschool
export PKG_CONFIG_PATH="/path/to/conda-environment/lib/pkgconfig"
export LD_LIBRARY_PATH="/path/to/conda-environment/lib:$LD_LIBRARY_PATH"
export LIBRARY_PATH="/path/to/conda-environment/lib:$LIBRARY_PATH"
export C_INCLUDE_PATH="/path/to/conda-environment/include:/path/to/conda-environment/include/python3.6m"
export CPLUS_INCLUDE_PATH="/path/to/conda-environment/include:/path/to/conda-environment/include/python3.6m"
```
## Install dependencies
- We need boost>=1.65. Installing something older than 1.66 leads to problems (see #23).
- [BOOST_PYTHON3_POSTFIX](https://github.com/openai/roboschool/blob/master/roboschool/cpp-household/Makefile#L59) was not identified properly. When installing boost with conda we have `BOOST_PYTHON3_POSTFIX=3`
- Having multiple versions of Boost installed is troublesome. Static linking of a (manually `-fPIC` compiled) boost-python library worked for me in this case.
```
conda install -c conda-forge boost
conda install -c flyem-forge assimp
conda install -c conda-forge pyqt ffmpeg
```

Then follow instructions in README to compile bullet3 and compile/install roboschool.

neutronest, 2018-03-13:
Maybe we also need run 
``` bash
conda install -c meznom boost-python 
```
?

nrontsis, 2018-03-13:
I believe `conda install -c conda-forge boost` install the whole boost, which includes the "boost-python" part. What is the error you get?

neutronest, 2018-03-14:
Have you ever encountered with this error https://github.com/openai/roboschool/issues/117 ? I'm now tring to fix it... It's sounds like there is something wrong with boost install step with conda. Actually I'm not sure if `conda install boost-python` is necessary. It's just a try.

isaacsultan, 2018-03-29:
I was receiving segmentation fault before trying @nrontsis suggestions.
Now I am receiving: 

``Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/kh/d2l11jtn55x7r9jdjhspvj0h0000gn/T/pip-build-x082wdas/unroll/
You are using pip version 9.0.1, however version 9.0.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
(py36) Isaacs-MacBook:~ isaacsultan$ pip3 install -e $ROBOSCHOOL_PATH
Obtaining file:///Users/isaacsultan/Code/roboschool
    Complete output from command python setup.py egg_info:
    Please follow instructions in README to build local (not global for your system) Bullet installation.
``
Did anybody experience this?

sumitsk, 2018-03-31:
I followed your instructions step by step, however, on the last step while executing: `pip install -e $ROBOSCHOOL_PATH`, I get the following error:
```
/usr/bin/ld: cannot find -lboost_python-py35
collect2: error: ld returned 1 exit status
Makefile:125: recipe for target '../cpp_household.so' failed
make: *** [../cpp_household.so] Error 1
```
I am installing it on a remote linux server with Ubuntu 16.04 via ssh. Conda version is 4.4.10 and python version is 3.6.

I can confirm that I have set up the relevant environment variables correctly:
`$ echo $LIBRARY_PATH` 
`/home/sumit/anaconda3/envs/drl/lib:`
`$ echo $LD_LIBRARY_PATH`
`/home/sumit/anaconda3/envs/drl/lib:`

I tried to find for boost_python library and got following results on executing 
`find . -name '*boost_python*'` in my home directory:
```
./anaconda3/envs/drl/lib/libboost_python-mt.a
./anaconda3/envs/drl/lib/libboost_python3.so.1.66.0
./anaconda3/envs/drl/lib/libboost_python-mt.so.1.55.0
./anaconda3/envs/drl/lib/libboost_python-mt.so
./anaconda3/envs/drl/lib/libboost_python3.a
./anaconda3/envs/drl/lib/libboost_python3.so
./anaconda3/pkgs/boost-python-1.55.0-1/lib/libboost_python-mt.a
./anaconda3/pkgs/boost-python-1.55.0-1/lib/libboost_python-mt.so.1.55.0
./anaconda3/pkgs/boost-python-1.55.0-1/lib/libboost_python-mt.so
./anaconda3/pkgs/boost-1.66.0-py36_1/lib/libboost_python3.so.1.66.0
./anaconda3/pkgs/boost-1.66.0-py36_1/lib/libboost_python3.a
./anaconda3/pkgs/boost-1.66.0-py36_1/lib/libboost_python3.so
```

I am not sure why am i encountering this issue. Any help is appreciated.

nrontsis, 2018-03-31:
First, `boost-python` package, as `boost` package is a superset of that.

Then, change [this](https://github.com/openai/roboschool/blob/master/roboschool/cpp-household/Makefile#L59) line to `BOOST_PYTHON = -lboost_python3`.

sumitsk, 2018-03-31:
Thanks @nrontsis . I have been able to successfully install roboschool now. However, I am stuck at a new problem. When I am resetting an environment, I am getting this error:
```
ArgumentError                             Traceback (most recent call last)
<ipython-input-4-c5672cf456ee> in <module>()
----> 1 env.reset()

~/anaconda3/envs/drl/lib/python3.6/site-packages/gym/wrappers/time_limit.py in reset(self)
     42         self._episode_started_at = time.time()
     43         self._elapsed_steps = 0
---> 44         return self.env.reset()

~/roboschool/roboschool/gym_mujoco_xml_env.py in _reset(self)
     36     def _reset(self):
     37         if self.scene is None:
---> 38             self.scene = self.create_single_player_scene()
     39         if not self.scene.multiplayer:
     40             self.scene.episode_restart()

~/roboschool/roboschool/gym_forward_walker.py in create_single_player_scene(self)
     20 
     21     def create_single_player_scene(self):
---> 22         return SinglePlayerStadiumScene(gravity=9.8, timestep=0.0165/4, frame_skip=4)
     23 
     24     def robot_specific_reset(self):

~/roboschool/roboschool/scene_abstract.py in __init__(self, gravity, timestep, frame_skip)
     23         self.dt = self.timestep * self.frame_skip
     24         self.cpp_world = cpp_household.World(gravity, timestep)
---> 25         self.cpp_world.set_glsl_path(os.path.join(os.path.dirname(__file__), "cpp-household/glsl"))
     26 
     27         self.big_caption = self.cpp_world.test_window_big_caption  # that's a function you can call

ArgumentError: Python argument types in
    World.set_glsl_path(World, str)
did not match C++ signature:
    set_glsl_path(World {lvalue}, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)
```

I can see that this error has been reported previously #23 , has anyone fixed it though?

Kelym, 2018-05-09:
Thank you! I was running Ubuntu 16.04, anaconda3 and got error complained about pkg_config, after exporting all the paths listed I was able to install roboschool.

rohansaphal97, 2018-05-09:
@sumitsk ..have you been able to resolve the last error you got.
I am getting the same thing.

sumitsk, 2018-05-12:
@rohansaphal97 . Unfortunately no! I tried installing roboschool on two linux machines one with GPU and other without. Roboschool is working just fine on the machine without GPU but not on the other one. Moreover, this problem occurs when you setup in a conda environment. Without it, installation works just fine.