openai/roboschool #124: why not use the official bullet3 repo?

mariusa, 2018-03-16:
Install notes point to
https://github.com/olegklimov/bullet3

but the official repo, up to date, is at
https://github.com/bulletphysics/bullet3

Why not use the official repo? If that has issues, what can we do to fix them?

benelot, 2018-04-28:
There is some hack Oleg introduced to the bullet3 repository but it is not in the official bullet repository because it messes heavily with the code. Our intent with pybullet-gym and the versions in the official bullet repository is to properly replicate MuJoCo envs with pybullet, and using the official bullet code for everything. As you can see, roboschool is already aging...

benelot, 2018-04-28:
In case you are interested in helping us with the official pybullet gym replication, you leave me a note.