openai/roboschool #126: missing bullet/PhysicsClientC_API.h

mariusa, 2018-03-16:
```
$ pip3 install -e $ROBOSCHOOL_PATH
```

gives
```
In file included from random-world-tools.cpp:1:0:
    household.h:3:10: fatal error: bullet/PhysicsClientC_API.h: No such file or directory
     #include <bullet/PhysicsClientC_API.h>
```

bullet local install has only these files:
```
$ ls $ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install/include/bullet/
btBulletCollisionCommon.h  Bullet2FileLoader  Bullet3Common    Bullet3Geometry  BulletCollision  BulletInverseDynamics  LinearMath
btBulletDynamicsCommon.h   Bullet3Collision   Bullet3Dynamics  Bullet3OpenCL    BulletDynamics   BulletSoftBody
```

How to fix this?

Thanks

yf-yang, 2018-09-15:
I've encountered the same problem,  just clone bullet3 from branch roboschool_self_collision

nsbb, 2019-06-17:
same problem. I tried even 
`sudo apt-get install libbullet-dev libbullet-doc libbulletml-dev`
but still same error.