openai/roboschool #130: Can you please create a docker image for OSX and Linux

FitMachineLearning, 2018-04-02:
Hi,

It would be great if a docker image could be created for those two environments.
It would 
1) significantly increase adoption for RoboSchool
2) reduce support  calls/ issues related to deployment, installs, and dev machine related configuration deviations.

Fit

pzhokhov, 2019-02-16:
@doviettung96's Dockerfile for now lives in a branch https://github.com/openai/roboschool/tree/pr168.
With the release of binary wheels the default installation process is a simple as `pip install roboschool` (provided there is a binary wheel for your system config), so Dockerfiles seem unnecessary. Closing this.

closed 2019-02-16 by pzhokhov

doviettung96, 2019-02-17:
Yeah. I have also realized that the binary installation is better. I have created the Dockerfile before it was released. For now, it is unnecessary. Thanks.