openai/roboschool #131: OpenGL version mismatch

cianci, 2018-04-04:
Okay, I am certain that this is entirely due to an embarrassingly simple error that I've made somewhere, but as I still can't see it, I must depend upon the kindness of strangers to point out where I've gone wrong...

Attempting to follow the setup instructions in the readme, on a machine with the following characteristics:

```
➜  ~ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.4 LTS"
➜  ~ nvidia-smi
Wed Apr  4 12:12:51 2018       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 390.30                 Driver Version: 390.30                    |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX TIT...  Off  | 00000000:03:00.0  On |                  N/A |
| 22%   40C    P8    15W / 250W |     31MiB / 12211MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+                                                                             +-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      5488      G   /usr/lib/xorg/Xorg                            28MiB |
+-----------------------------------------------------------------------------+
➜  ~ cat /var/log/Xorg.0.log | grep glx -ni -C5 
52-
53-[ 76722.806] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
54-[ 76722.806] (II) xfree86: Adding drm device (/dev/dri/card0)
55-[ 76722.807] (--) PCI:*(0:3:0:0) 10de:17c2:3842:2990 rev 161, Mem @ 0xfa000000/16777216, 0xc0000000/268435456, 0xd0000000/33554432, I/O @ 0x0000e000/128, BIOS @ 0x????????/524288
56-[ 76722.808] (WW) "glamoregl" will not be loaded unless you've specified it to be loaded elsewhere.
57:[ 76722.808] (II) "glx" will be loaded by default.
58:[ 76722.808] (II) LoadModule: "glx"
59:[ 76722.808] (II) Loading /usr/lib/x86_64-linux-gnu/xorg/extra-modules/libglx.so
60:[ 76722.812] (II) Module glx: vendor="NVIDIA Corporation"
61-[ 76722.812] 	compiled for 4.0.2, module version = 1.0.0
62-[ 76722.812] 	Module class: X.Org Server Extension
63:[ 76722.812] (II) NVIDIA GLX Module  390.30  Wed Jan 31 21:27:29 PST 2018
64-[ 76722.812] (==) Matched nvidia as autoconfigured driver 0
65-[ 76722.812] (==) Matched nouveau as autoconfigured driver 1
66-[ 76722.812] (==) Matched nvidia as autoconfigured driver 2
67-[ 76722.812] (==) Matched nouveau as autoconfigured driver 3
68-[ 76722.812] (==) Matched modesetting as autoconfigured driver 4
--
260-[ 76723.495] (II) Module "dri2" already built-in
261-[ 76723.495] (II) NVIDIA(0): [DRI2] Setup complete
262-[ 76723.495] (II) NVIDIA(0): [DRI2]   VDPAU driver: nvidia
263-[ 76723.495] (--) RandR disabled
264-[ 76723.497] (II) SELinux: Disabled on system
265:[ 76723.497] (II) Initializing extension GLX
266:[ 76723.497] (II) Indirect GLX disabled.
267-[ 76723.526] (II) config/udev: Adding input device Power Button (/dev/input/event1)
268-[ 76723.527] (**) Power Button: Applying InputClass "evdev keyboard catchall"
269-[ 76723.527] (II) LoadModule: "evdev"
270-[ 76723.527] (II) Loading /usr/lib/xorg/modules/input/evdev_drv.so
271-[ 76723.527] (II) Module evdev: vendor="X.Org Foundation"
```

And yet, somehow when I run the test:

```
➜  ~ python3 $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: Environment '<class 'roboschool.gym_humanoid_flagrun.RoboschoolHumanoidFlagrun'>' has deprecated methods '_step' and '_reset' rather than 'step' and 'reset'. Compatibility code invoked. Set _gym_disable_underscore_compat = True to disable this behavior.
Cannot initialize OpenGL context.
Requested version: 4.1
Actual version: 1.2
(it must be at least 3.3 to work)
For possible fixes, see:
https://github.com/openai/roboschool/issues/2
python3: render-simple.cpp:695: void SimpleRender::opengl_init(const boost::shared_ptr<Household::World>&): Assertion `0' failed.
[1]    16633 abort (core dumped)  python3 $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
```

And this perplexes me, as I believe the OpenGL version on this Titan X card to be 4.4; how do I go about determining where the 1.2 is coming from, so that I can expunge it?

Apologies in advance for whatever information I should have provided here but didn't; happy to oblige any further data requests.

sumitsk, 2018-04-05:
I am not an expert in this area but these are the things I would try to debug the issue. 

Try `locate -r /gl.h$` first to see all the places which contains gl.h file. 
Then, `glxinfo | grep version` to check the version of opengl. 
If still no luck, then obviously you can `trace` and `strace` to determine the exact path of the module.

ouyangzhuzhu, 2018-11-24:
I meet  the  same  problem  and  add from OpenGL import GLU to the beginning of python main did not work for me............:  sad,,,,,,,,,,,,

Cannot initialize OpenGL context.
Requested version: 4.1
Actual version: 1.4
(it must be at least 3.3 to work)
For possible fixes, see:

https://github.com/openai/roboschool/issues/2

python3: render-simple.cpp:695: void SimpleRender::opengl_init(const boost::shared_ptr<Household::World>&): Assertion `0' failed.
Aborted (core dumped)