openai/roboschool #133: ld: library not found for -lboost_python3

wsgdrfz, 2018-04-15:
When I get into the final step
`pip3 install -e $ROBOSCHOOL_PATH`

Faults:
```
gcc -shared -o../cpp_household.so .build-release/physics-bullet.o .build-release/assets-mesh.o .build-release/random-world-tools.o .build-release/render-glwidget.o .build-release/render-hud.o .build-release/render-simple.o .build-release/render-simple-primitives.o .build-release/python-binding.o -framework OpenGL -lstdc++ `pkg-config --libs Qt5OpenGL Qt5Widgets assimp python-3.6`  -Lbullet_local_install/lib -lLinearMath -lBullet3Common -lBulletCollision -lBulletDynamics -lBulletInverseDynamics -lPhysicsClientC_API -lboost_python3
    ld: library not found for -lboost_python3
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make: *** [../cpp_household.so] Error 1
    cd /Users/wsgdrfz/Downloads/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so

    C++ dependencies for this project are:

    bullet
    tinyxml
    boost_python
    assimp
    Qt5

    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
```

I think it might because the formal operation:
`brew reinstall boost-python --without-python --with-python3 --build-from-source
`
I got the warning:
```
Warning: boost-python: this formula has no --with-python3 option so it will be ignored!
Warning: boost-python: this formula has no --without-python option so it will be ignored!
```

wsgdrfz, 2018-04-16:
Okay, I found the solution.

Just use:
`brew reinstall boost-python3 --with-python --build-from-source`

closed 2018-04-16 by wsgdrfz

juliayukam, 2018-04-20:
I am running into the same problem as posted above. ( I am trying to compile caffe)

I tried the posted method by OP of using 
brew reinstall boost-python3 --with-python --build-from-source

and after a successful reinstall, I am still getting the error message:
ld: library not found for -lboost_python3
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make: *** [../cpp_household.so] Error 1

How can I find -lboost_python3 file and specify the path to the makefile?
(Or... what is an alternative?)
Thank you

hammadullah125, 2018-06-21:
I'm having the same exact problem. Unable to install the boost-python. Any solutions???

Likho, 2018-08-04:
@wsgdrfz solution worked for  me, thank you!!

GoingMyWay, 2018-08-18:
@juliayukam  @hammadullah125 

I installed boost.python on MacOS with `brew install boost-python --with-python3` and I got the same error when compiling my code.

```
ld: library not found for -lboost_python3
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

And I  found I installed the boost-python in 

```
/usr/local/Cellar/boost-python3/1.67.0_1/lib
``` 

in this directory, I got

```
libboost_numpy37-mt.a      libboost_numpy37.a         libboost_python37-mt.dylib libboost_python37.dylib
libboost_numpy37-mt.dylib  libboost_numpy37.dylib     libboost_python37-mt.a     libboost_python37.a
```

And you can make a soft link by 

```
sudo ln -s libboost_python37.dylib libboost_python3.dylib
```

So that you can get `libboost_python3.dylib` and this issue is fixed. Hope it will help you fix this issue.