openai/roboschool #136: ld: library not found for -lboost_python3

raymondchua, 2018-05-02:
Hi, 
I have an error during installation and it reports that library not found for -lboost_python3.
Any ideas on how I can fix it? I tried to install in my conda environment.

```
Mac, assumed python3, homebrew branch
Link against python 3.5
Slow hardware or software render (no shadows)
rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so ../cpp_household_d.so  .build-release/physics-bullet.o  .build-release/assets-mesh.o  .build-release/random-world-tools.o  .build-release/render-glwidget.o  .build-release/render-hud.o  .build-release/render-simple.o  .build-release/render-simple-primitives.o  .build-release/test-tool-qt4.o  .build-release/python-binding.o  .build-debug/physics-bullet.o  .build-debug/assets-mesh.o  .build-debug/random-world-tools.o  .build-debug/render-glwidget.o  .build-debug/render-hud.o  .build-debug/render-simple.o  .build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o  .build-debug/python-binding.o .generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep .build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep .build-release/render-glwidget.o.dep .build-release/render-hud.o.dep .build-release/render-simple.o.dep .build-release/render-simple-primitives.o.dep .build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep .build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep .build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep .build-debug/render-hud.o.dep .build-debug/render-simple.o.dep .build-debug/render-simple-primitives.o.dep .build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep
rm -rf .generated
rm -rf .build-debug
rm -rf .build-release
Mac, assumed python3, homebrew branch
Link against python 3.5
Slow hardware or software render (no shadows)
mkdir -p .generated
mkdir -p .build-release
mkdir -p .build-debug
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-glwidget.cpp  -o.build-release/render-glwidget.o -MMD -MF .build-release/render-glwidget.o.dep
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-hud.cpp  -o.build-release/render-hud.o -MMD -MF .build-release/render-hud.o.dep
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-simple.cpp  -o.build-release/render-simple.o -MMD -MF .build-release/render-simple.o.dep
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-simple-primitives.cpp  -o.build-release/render-simple-primitives.o -MMD -MF .build-release/render-simple-primitives.o.dep
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c python-binding.cpp  -o.build-release/python-binding.o -MMD -MF .build-release/python-binding.o.dep
In file included from python-binding.cpp:1:
In file included from /usr/local/include/boost/python.hpp:11:
In file included from /usr/local/include/boost/python/args.hpp:10:
In file included from /usr/local/include/boost/python/args_fwd.hpp:10:
In file included from /usr/local/include/boost/python/handle.hpp:10:
/usr/local/include/boost/python/cast.hpp:73:20: warning: unused typedef 'must_be_a_complete_type'
      [-Wunused-local-typedef]
      typedef char must_be_a_complete_type[sizeof(T)];
                   ^
In file included from python-binding.cpp:1:
In file included from /usr/local/include/boost/python.hpp:18:
In file included from /usr/local/include/boost/python/class.hpp:17:
In file included from /usr/local/include/boost/python/data_members.hpp:15:
/usr/local/include/boost/python/make_function.hpp:58:32: warning: unused typedef 'assertion'
      [-Wunused-local-typedef]
          >::too_many_keywords assertion;
                               ^
In file included from python-binding.cpp:1:
In file included from /usr/local/include/boost/python.hpp:18:
In file included from /usr/local/include/boost/python/class.hpp:20:
/usr/local/include/boost/python/init.hpp:227:34: warning: unused typedef 'assertion'
      [-Wunused-local-typedef]
            >::too_many_keywords assertion;
                                 ^
/usr/local/include/boost/python/init.hpp:236:34: warning: unused typedef 'assertion'
      [-Wunused-local-typedef]
            >::too_many_keywords assertion;
                                 ^
In file included from python-binding.cpp:1:
In file included from /usr/local/include/boost/python.hpp:18:
In file included from /usr/local/include/boost/python/class.hpp:24:
/usr/local/include/boost/python/object/pickle_support.hpp:110:31: warning: unused typedef
      'error_type' [-Wunused-local-typedef]
          Class_>::error_type error_type;
                              ^
In file included from python-binding.cpp:1:
In file included from /usr/local/include/boost/python.hpp:18:
/usr/local/include/boost/python/class.hpp:145:81: warning: unused typedef 'test2'
      [-Wunused-local-typedef]
            typedef typename assertion<is_member_function_pointer<Fn> >::failed test2;
                                                                                ^
/usr/local/include/boost/python/class.hpp:143:68: warning: unused typedef 'test1'
      [-Wunused-local-typedef]
            typedef typename assertion<is_polymorphic<T> >::failed test1;
[... 5 lines cut ...]
          >::type assertion;
                  ^
In file included from python-binding.cpp:1:
In file included from /usr/local/include/boost/python.hpp:41:
/usr/local/include/boost/python/make_constructor.hpp:177:32: warning: unused typedef 'assertion'
      [-Wunused-local-typedef]
          >::too_many_keywords assertion;
                               ^
9 warnings generated.
[... cut ...] arMath -lBullet3Common -lBulletCollision -lBulletDynamics -lBulletInverseDynamics -lPhysicsClientC_API -lboost_python3
ld: library not found for -lboost_python3
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [../cpp_household.so] Error 1

C++ dependencies for this project are:

bullet
tinyxml
boost_python
assimp
Qt5

If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
Install dependencies that pkg-config cannot find.
```

olegklimov, 2018-05-02:
Oh just run `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` and see if you can find boost_python3 in libs yourself. Can be 35 on the end or something like that, or path is wrong.

raymondchua, 2018-05-02:
Hi @olegklimov,
Thanks for the prompt reply. 
This is what I get:
```
pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5
-DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/local/Cellar/qt/5.10.1/include/QtOpenGL -I/usr/local/Cellar/qt/5.10.1/include -I/usr/local/Cellar/qt/5.10.1/include/QtWidgets -I/usr/local/Cellar/qt/5.10.1/include -I/usr/local/Cellar/qt/5.10.1/include/QtGui -I/usr/local/Cellar/qt/5.10.1/include -I/usr/local/Cellar/qt/5.10.1/include/QtCore -I/usr/local/Cellar/qt/5.10.1/include -I/Users/raymondchua/miniconda3/envs/cvpr352/include/python3.5m -I/usr/local/Cellar/assimp/4.1.0/include
```