openai/roboschool #141: Segmentation fault when multiple rendering

neka-nat, 2018-06-03:
Hi,
#73 issue is occurring again.
In addition, it seems that rendering of multiple environment like the following script will not work yet.
Is there a way to solve something?

```
import gym, roboschool
import time

env = [None, None]
for i in range(2):
    env[i] = gym.make('RoboschoolReacher-v1')
    env[i].reset()

while True:
    for i in range(2):
        action = env[i].action_space.sample()
        observation, reward, done, info = env[i].step(action)
        print("Render")
        env[i].render()
    time.sleep(1)
```

ashigirl96, 2018-06-14:
I have the similar problem. 

```
> gdb python
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Registered pretty printers for UE4 classes
Reading symbols from python...done.
(gdb) run dynamics.py
Starting program: /home/username/.anaconda3/envs/roboschool/bin/python dynamics.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff097f700 (LWP 20787)]
[New Thread 0x7ffff017e700 (LWP 20788)]
[New Thread 0x7fffeb97d700 (LWP 20789)]
[New Thread 0x7fffe917c700 (LWP 20790)]
[New Thread 0x7fffe697b700 (LWP 20791)]
[Thread 0x7fffe697b700 (LWP 20791) exited]
[Thread 0x7fffe917c700 (LWP 20790) exited]
[Thread 0x7fffeb97d700 (LWP 20789) exited]
[Thread 0x7ffff017e700 (LWP 20788) exited]
[Thread 0x7ffff097f700 (LWP 20787) exited]
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: Environment '<class 'roboschool.gym_mujoco_walkers.RoboschoolAnt'>' has deprecated methods '_step' and '_reset' rather than 'step' and 'reset'. Compatibility code invoked. Set _gym_disable_underscore_compat = True to disable this behavior.
[New Thread 0x7fffe697b700 (LWP 20796)]

Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007fff83be36cc in QGLContext::contextHandle() const () from /home/username/.anaconda3/envs/roboschool/bin/../lib/libQt5OpenGL.so.5
```

olegklimov, 2018-06-14:
I can confirm this is happening.