openai/roboschool #142: Half cheetah state space not defined properly

rohansaphal97, 2018-06-15:
I have some questions regarding the roboschool Half-Cheetah.

1.) I see that the observation space for Half-Cheetah is 26. Can anyone tell me what is each value for?- I only counted 18. (also, some of the values seem to remain 0 for all timesteps)

2.) In the half_cheetah.xml under roboschool/mujoco_assets, there is the following comment:

Cheetah Model
[image]

Could you please confirm to me if the order presented here is the same with the order they appear in the observation matrix?
Also the last 8 values of the 26 values are not always zero. They are either 0 and one.

Please help.

Thank you

rohansaphal97, 2018-06-15:
[half cheetah.xlsx](https://github.com/openai/roboschool/files/2107549/half.cheetah.xlsx)

this is the state space for 100 timesteps. It has 26 columns corresponding to the 26 states. If you notice, column 2 and 3 are always 0 and 1 respectively. column 5 and column 6 are always zero. Thats why there is a lot of confusion.

Please explain

olegklimov, 2018-06-16:
I guess it's upper part of the legs contact with the ground. Please take a look at the source code.

rohansaphal97, 2018-06-16:
i found only these two codes
https://github.com/openai/roboschool/blob/master/roboschool/mujoco_assets/half_cheetah.xml

https://github.com/openai/roboschool/blob/master/roboschool/gym_mujoco_xml_env.py

are these the ones you ae talking about?..i wasnt able to find any useful info from these.

There are 10 columns with zero and ones. There are only 16 columns remaining but 18 states are given in the xml files. How does this work? ( x,y,z and velocities,6 joints and their angular velocities--> 18 values)

olegklimov, 2018-06-16:
https://github.com/openai/roboschool/blob/master/roboschool/gym_forward_walker.py#L45

alexvicegrab, 2020-09-25:
Unfortunately, the documentation is, erm... lacking. It should be possible to understand the mapping from the original Half-Cheetah to this implementation from a basic DocString in the code.

i.e. are all the other state-space elements the same as in Mujoco, or are they different? To someone who is not intimately familiar with the code and the problem, that is not obvious, and increases the threshold to usage.