openai/roboschool #149: /usr/bin/ld: cannot find -lboost_python-py36

huangjiancong1, 2018-08-15:
system: Ubuntu16.04
virtualenv create the only python3.6 environment:

`$ pip3 install -e $ROBOSCHOOL_PATH`

error output:
```
    gcc -shared -o../cpp_household.so .build-release/physics-bullet.o .build-release/assets-mesh.o .build-release/random-world-tools.o .build-release/render-glwidget.o .build-release/render-hud.o .build-release/render-simple.o .build-release/render-simple-primitives.o .build-release/python-binding.o -L/usr/lib64 -lm -lGL -lGLU -lstdc++ `pkg-config --libs Qt5OpenGL Qt5Widgets assimp python-3.6` -Wl,-z,origin -Wl,--disable-new-dtags -Wl,-rpath,'$ORIGIN/cpp-household/bullet_local_install/lib' -Lbullet_local_install/lib -lLinearMath -lBullet3Common -lBulletCollision -lBulletDynamics -lBulletInverseDynamics -lPhysicsClientC_API -lboost_python-py36
    /usr/bin/ld: cannot find -lboost_python-py36
    collect2: error: ld returned 1 exit status
    Makefile:125: recipe for target '../cpp_household.so' failed
    make: *** [../cpp_household.so] Error 1
    cd /home/huang/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
```

I don't know what going on, i can test [boost example](https://www.boost.org/doc/libs/1_63_0/libs/python/doc/html/tutorial/index.html) under python3.6 environment.
Can we due with this question together?

huangjiancong1, 2018-08-15:
It seem that there have no `libboost_python-py36` generated when i install Boost.  But generated `libboost_python-py35` under the python3.6 only virtual environment.

John1231983, 2018-12-20:
@huangjiancong1 : How do you solve your problem?

zanovis, 2019-04-06:
@id9502's fix partially worked. 

I replaced BOOST_PYTHON3_POSTFIX in line:59 with "3" and it fixed the issue that `/usr/bin/ld: cannot find -lboost_python36`

This is probably not a sustainable fix, so if someone has a better suggestion please let us know.

vineetsagar7, 2019-04-18:
hon3.6m -ldl -framework CoreFoundation -o libcavehull.so
ld: library not found for -lboost_python-py36
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [libcavehull.so] Error 1

Getting the above error when executing "make" on mac.