openai/roboschool #151: Apply pure force/torque, and compute acceleration.

charles-blouin, 2018-10-04:
I would like to use the roboschool framework for applied robotics. Specifically, for a quadcopter simulation. In practice, I will have access to the raw accelerometer and gyroscope data. Is there a way to obtain the accelerations without differentiating the position and orientation? Would additional python binding be needed?

Also, is there the equivalent of Mujoco qfrc_applied to apply a force and a torque on an object?

Thanks!

olegklimov, 2018-10-04:
That seems to be a very simple environment. I'd recommend pybullet maybe for doing this, because you have immediate access to simulation internals from python.

closed 2018-10-04 by olegklimov

charles-blouin, 2018-10-04:
Thanks. After looking into it, seems to make more sense to use pybullet here.
The pybullet function for applying a force is applyExternalForce. To obtain acceleration, it seems that the only way now is to differentiate the velocity.