openai/roboschool #152: Hopper alive_bonus() pitch condition too strict?

cpacker, 2018-10-06:
It seems that `abs(pitch) < 1.0` in the following code is causing some episodes in Hopper to terminate (subjectively) early:

https://github.com/openai/roboschool/blob/e17139b87033b1ec9ca1ca3a92eeb60ccd9e7014/roboschool/gym_mujoco_walkers.py#L18-19

In the following video the episode appears to end before Hopper has reached a "fatal" state (at least relative to HalfCheetah where it is clear that the agent has fallen before ending):

https://drive.google.com/open?id=1md5ieWJE4cX6VBGYCOBVcpJTHd2typY_

Slightly relaxing the condition to `abs(pitch) < 1.1` (https://github.com/cpacker/roboschool/commit/8eacb67586a23a8d1ef3adc9d05f62adb1c94743) appears to fix the problem (at least I didn't notice it in any subsequent recordings). Same trained agent with the same environment settings, but with modified pitch condition:

https://drive.google.com/open?id=1Jd2qsegc9sAULtEexlrOwpshYA1kMwnx

olegklimov, 2018-10-06:
I guess you're right! But also it doesn't matter much, because the agent will notice episode ends with certain states and will avoid it. It's just definition of a task. I've seen that before, once someone proposed a change to CartPole in gym (the most basic classic environment) better integration method. We decided in that case maybe integration method is better, but it doesn't worth it to change the task, increase version number. Nice video!

closed 2018-10-06 by olegklimov

cpacker, 2018-10-06:
@olegklimov I agree that it doesn't make sense to change the task definition for Hopper-v0/v1 (ie change the `abs(pitch) < 1.0` to `abs(pitch) < 1.1` in the current codebase), but I do think it's something important to consider for a future version bump (Hopper-v2, or in "roboschool2", if the same Hopper is included).