openai/roboschool #158: Complete install guidelines?

JorDikk, 2018-11-13:
Are the installation guidlines complete? 
I tried on 2 different Ubuntu machines and I get the same error:
Directory '/path/to/roboschool' is not installable. File 'setup.py' not found.

These were my commands in terminal:

ROBOSCHOOL_PATH=/path/to/roboschool

apt install cmake ffmpeg pkg-config qtbase5-dev libqt5opengl5-dev libassimp-dev libpython3.5-dev libboost-python-dev libtinyxml-dev

git clone https://github.com/olegklimov/bullet3 -b roboschool_self_collision

mkdir bullet3/build

cd    bullet3/build

cmake -DBUILD_SHARED_LIBS=ON -DUSE_DOUBLE_PRECISION=1 -DCMAKE_INSTALL_PREFIX:PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install -DBUILD_CPU_DEMOS=OFF -DBUILD_BULLET2_DEMOS=OFF -DBUILD_EXTRAS=OFF  -DBUILD_UNIT_TESTS=OFF -DBUILD_CLSOCKET=OFF -DBUILD_ENET=OFF -DBUILD_OPENGL3_DEMOS=OFF ..

make -j4

make install

cd ../..

pip3 install -e $ROBOSCHOOL_PATH

Are there other things I should do in between the commands?
Or anyone knows how to fix this error?
Thx

olegklimov, 2018-11-13:
Replace /path/to/roboschool with a folder you can write to.

closed 2018-11-13 by olegklimov

JorDikk, 2018-11-13:
What would that folder be? I tried my Documents folder and also made another folder and tried it. Kept getting the same error (but than for that particular folder)