openai/roboschool #167: Installation Problem

japneet644, 2019-01-03:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version (3.10.2-1ubuntu2).
ffmpeg is already the newest version (7:3.4.2-2).
libpython3.5-dev is already the newest version (3.5.6-1+bionic1).
libpython3.5-dev set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libboost-python-dev : Depends: libboost-python1.65-dev but it is not going to be installed
 qtbase5-dev : Depends: libgl1-mesa-dev but it is not going to be installed or
                        libgl-dev
               Depends: libgles2-mesa-dev but it is not going to be installed or
                        libgles2-dev but it is not installable
               Depends: libglu1-mesa-dev but it is not going to be installed or
                        libglu-dev
               Depends: libxext-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
I am beginner in RL. during installation I am getting the above error when i run
`sudo apt install cmake ffmpeg pkg-config qtbase5-dev libqt5opengl5-dev libassimp-dev libpython3.5-dev libboost-python-dev libtinyxml-dev`
Please help me. I am already fed up of this installation procedure

doviettung96, 2019-01-30:
Hi @japneet644 ,
Have you managed to issue that problem?
I think it will be easier to use Docker to install Roboschool as well as other packages. 
If you would like, I will try to submit a Pull request with the installation of Roboschool + other packages (it depends on your need).
Thanks.

pzhokhov, 2019-02-27:
the binary wheel should work for ubuntu 18.04; so the installation should now be as simple as `pip install roboschool`. Closing this.

closed 2019-02-27 by pzhokhov

hzm2016, 2019-09-24:
> Hi @japneet644 ,
> Have you managed to issue that problem?
> I think it will be easier to use Docker to install Roboschool as well as other packages.
> If you would like, I will try to submit a Pull request with the installation of Roboschool + other packages (it depends on your need).
> Thanks.

I have the same issue when i install roboschool by pip install, like "ImportError: cannot import name 'cpp_household'"

pzhokhov, 2019-09-24:
@hzm2016  could you specify OS version, and roboschool version (e.g. output of `pip freeze | grep roboschool`)?

hzm2016, 2019-09-25:
> @hzm2016 could you specify OS version, and roboschool version (e.g. output of `pip freeze | grep roboschool`)?

thanks a lot, 
my cluster version HP Xeon two sockets 12-Core 64-bit Linux cluster, CentOS 6, and roboschool version==1.0.48

pzhokhov, 2019-09-25:
ooofff roboschool wheels are not available for centos :( You can try to compile it from source, as described here: https://github.com/openai/roboschool#installation-from-source

hzm2016, 2019-09-25:
> ooofff roboschool wheels are not available for centos :( You can try to compile it from source, as described here: https://github.com/openai/roboschool#installation-from-source

@pzhokhov , I have tried, folllow this .sh
apt install unzip cmake ffmpeg pkg-config qtbase5-dev libqt5opengl5-dev libassimp-dev libpython3.5-dev libboost-python-dev libtinyxml-dev
git clone https://github.com/openai/roboschool.git roboschool_lib
git clone https://github.com/olegklimov/bullet3 -b roboschool_self_collision
export ROBOSCHOOL_PATH=/workspace/roboschool_lib
mkdir bullet3/build
cd    bullet3/build
cmake -DBUILD_SHARED_LIBS=ON -DUSE_DOUBLE_PRECISION=1 -DCMAKE_INSTALL_PREFIX:PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install -DBUILD_CPU_DEMOS=OFF -DBUILD_BULLET2_DEMOS=OFF -DBUILD_EXTRAS=OFF  -DBUILD_UNIT_TESTS=OFF -DBUILD_CLSOCKET=OFF -DBUILD_ENET=OFF -DBUILD_OPENGL3_DEMOS=OFF ..
make -j4
make install
cd ../..
pip install -e $ROBOSCHOOL_PATH

but I met the same error. "cann't import name cpp_household"
[image]

pzhokhov, 2019-09-25:
I have been trying to make it work for a while, but with not a whole lot of success. Thing is, both bullet and qt5 require c++11 compiler, and by default, to my understanding, centos 6 ships with gcc 4.4 which is not c++11 compatible. While that does not mean installing roboschool on centos 6 is impossible, it is be rather painful. So far, I managed to compile gcc 4.8 on centos 6 docker image (whew, only took half a day) by loosely following the instructions here: https://wiki.qt.io/How_to_Install_Qt_5_and_Qwt_on_CentOS_6

pzhokhov, 2019-09-25:
however, qt5 with opengl support on centos 6 is something I have not figured out how to do. if you manage to install that on centos 6, the rest should fall in line :)