openai/roboschool #170: Roboschool build from source issues

christopherhesse, 2019-02-08:
From @olegklimov:

Some things I had to do to compile:
```
diff --git a/roboschool/cpp-household/Makefile b/roboschool/cpp-household/Makefile
index b16e76ba..435089fe 100644
--- a/roboschool/cpp-household/Makefile
+++ b/roboschool/cpp-household/Makefile
@@ -28,7 +28,7 @@ ifeq ($(UNAME),Linux)
     $(info Linux, python 2 branch)
     PYTHON = $(shell python -c "import sys; print('%i.%i' % sys.version_info[:2])")
   endif
-  RPATH=-Wl,-z,origin -Wl,--disable-new-dtags,-rpath,'$(BULLET_PATH)/lib',-rpath,'$(HOME)/.boost/lib'
+  RPATH=-Wl,-z,origin -Wl,--disable-new-dtags,-rpath,'$(BULLET_PATH)/lib',-rpath,'$(HOME)/.boost/lib' -L$(BULLET_PATH)/lib
 endif
 
 ifeq ($(UNAME),Darwin)
@@ -48,8 +48,8 @@ ifeq ($(UNAME),Darwin)
   RPATH = -Wl,-undefined,dynamic_lookup
 endif
 
-INC += `$(PKG) --cflags Qt5Widgets Qt5OpenGL assimp bullet`
-LIBS += `$(PKG) --libs Qt5Widgets Qt5OpenGL assimp bullet`
+INC += `$(PKG) --cflags Qt5Widgets Qt5OpenGL assimp bullet python3`
+LIBS += `$(PKG) --libs Qt5Widgets Qt5OpenGL assimp bullet python3`
 LIBS    += -lstdc++
 INC     += -I/usr/local/include/bullet
 LIBS    += $(RPATH) -lLinearMath -lBullet3Common -lBulletCollision -lBulletDynamics -lBulletInverseDynamics -lBulletSoftBody -lPhysicsClientC_API
diff --git a/roboschool/cpp-household/physics-bullet.cpp b/roboschool/cpp-household/physics-bullet.cpp
index 63c3935e..de77d235 100644
--- a/roboschool/cpp-household/physics-bullet.cpp
+++ b/roboschool/cpp-household/physics-bullet.cpp
@@ -95,7 +95,6 @@ shared_ptr<Robot> World::load_urdf(const std::string& fn, const btTransform& tr,
 std::list<shared_ptr<Robot>> World::load_sdf_mjcf(const std::string& fn, bool mjcf)
 {
        std::list<shared_ptr<Robot>> ret;
-       const int MAX_SDF_BODIES = 512;
        int bodyIndicesOut[MAX_SDF_BODIES];
        int N;
        if (mjcf) {
```

But that only works if I set BULLET_PATH:
```
BULLET_PATH=/home/me/roboschool/roboschool/cpp-household/bullet_local_install make -C roboschool/cpp-household/
```

Attempting to start:
```
gdb --args python3 agent_zoo/RoboschoolHopper_v0_2017may.py
...
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007fffdd451940 in btDiscreteDynamicsWorld::addConstraint(btTypedConstraint*, bool) () from /usr/local/lib/libBulletDynamics.so.2.88
```
Not sure what is going here. So I didn't manage to make it work so far. Maybe will attempt later.

pzhokhov, 2019-02-08:
1) correct, the `BULLET_PATH` needs to be set; however, this should happen when you `source exports.sh` (as per installation instructions in the updated README.md).
2) Segfault - what appears to have happened is linking against wrong version of bullet (global vs local - local should now be installed in `~/.forked_bullet`
So for both of those the answer is likely `source exports.sh` before running make. @olegklimov when you have a chance could you try again with that? Thanks!

doviettung96, 2019-02-09:
Did you fix that? I have just used roboschool and get the following message:
ImportError: cannot import name 'cpp_household'

christopherhesse, 2019-02-11:
@doviettung96 are you using the binary install or building from source?

pzhokhov, 2019-02-11:
@doviettung96 Could you provide more details? How was roboschool installed (i.e. from binary wheel or from source)? If installed from source, how did you install it? Which os are you using? 
Thanks!

doviettung96, 2019-02-14:
@christopherhesse  and @pzhokhov ,
I built it from source to a docker machine (ubuntu 16.04). The Dockerfile is as in #168 .
Previously, it was smoothly built. Recently, just after you open this issue, I could not install it anymore.
Thanks.

pzhokhov, 2019-02-15:
@doviettung96 I will look into what goes wrong in the Dockerfile you refer to. In the mean time, you can use binaries (we test them with 16.04, so I don't expect any surprises there)

doviettung96, 2019-02-15:
@pzhokhov, 
I will try it. The binary installation should be easier than from source one. Thank you.

doviettung96, 2019-02-15:
@pzhokhov ,
Thank you. I have tried the binary one and it succeeded.

pzhokhov, 2019-02-16:
@doviettung96 I think I fixed the Dockerfile that you referred to with installation from source; in this branch: https://github.com/openai/roboschool/tree/pr168. The problem was that previously we were using system-level libboost-python package; and now roboschool is switched to using compiled-from-source boost-python. The latter is necessary, for instance, if you want to use python3.6 on ubuntu 16.04 (system-level pacakge only provides libraries for 3.5). System-level boost and compiled from source differ a little bit in suffixes, and that's what was causing problems.
That being said,
1) the preferred method of usage is still binaries (because they install faster and contain all or almost all dependencies); I would only go the source code installation route if you want to modify some of the roboschool environments
2) I don't want to keep that Dockerfile in the master branch - now that we release binaries, creating a Dockerfile "just to make roboschool work" is unnecessary. As such, I'll close the referred PR.

doviettung96, 2019-02-17:
@pzhokhov ,
With the release of the binary installation, I think no one should have installation problems from now on. Thus, my docker installation is unnecessary.
Thank you.

xbit126, 2019-02-20:
> @pzhokhov ,
> With the release of the binary installation, I think no one should have installation problems from now on. Thus, my docker installation is unnecessary.
> Thank you.

I also install the roboschool, but when I `import roboschool`, I got 
ImportError: cannot import name 'cpp_household'
could you please add more details about binary installation? how was roboschool installed?

pzhokhov, 2019-02-21:
the binary wheels can be installed via `pip install roboschool` (note that the folder in which you are running the command should not contain subfolder `roboschool` with the file `setup.py` - otherwisem pip will get confused and will attempt installation from local sources instead. 
If the above command fails for you, please provide os version, and the sequence of commands you are running to reproduce the error. Thanks!

xht033, 2019-03-09:
ubuntu 18.04
conda 4.6.7
pip install roboschool
----------------------------
ImportError: cannot import name 'cpp_household'

toandx, 2019-05-03:
ubuntu 16.04
python 3.5.2
pip install roboschool 
AttributeError: 'NoneType' object has no attribute 'human_render_detected'
#187

pzhokhov, 2019-05-03:
@toandx moving discussion to the issue #187  (TLDR - you need to call `env.reset()` before using the environment

pzhokhov, 2019-05-03:
@xht033 if you have a chance, could you try again with latest gym and latest roboschool wheel? Thanks!

doviettung96, 2019-05-06:
@pzhokhov ,
Recently, as I tried to build roboschool from source in a Docker machine (ubuntu 16.04),
at the step "RUN ./docker_build_wheel.sh", I got the following error:

In file included from /root/.boost/include/boost/python/detail/prefix.hpp:13:0,
                 from /root/.boost/include/boost/python/args.hpp:8,
                 from /root/.boost/include/boost/python.hpp:11,
                 from python-binding.cpp:1:
/root/.boost/include/boost/python/detail/wrap_python.hpp:50:23: fatal error: pyconfig.h: No such file or directory
compilation terminated.
make: * [.build-release/python-binding.o] Error 1
make: * Waiting for unfinished jobs....

Do you have any advice for creating a Docker image containing Roboschool + OpenAi Baselines?
Thanks.

doviettung96, 2019-05-06:
Hi @pzhokhov ,
Recently, I have built roboschool from source on a Docker machine (ubuntu 16.04). However, as I tried to import roboschool, I got the following error:
ImportError: /roboschool/roboschool/cpp_household.so: undefined symbol: PyInt_Type
Any suggestion?
Thanks.

pzhokhov, 2019-05-06:
I'd guess some library version mismatch? I have tried master branch and the following Dockerfile:
```docker
FROM ubuntu:16.04
# get dependencies (ubuntu docker image is really stripped down, even curl and git are absent)
RUN apt-get -y update && apt-get -y install python3-dev python3-pip curl \
               apt-utils libgl1-mesa-dev qtbase5-dev \
               libqt5opengl5-dev libassimp-dev patchelf cmake pkg-config git 
# set python and pip to point to python3 versions
RUN bash -c "ln -s $(which python3) /usr/bin/python && ln -s $(which pip3) /usr/bin/pip"
COPY . /code
RUN  cd /code && \
     bash -c ". ./exports.sh && ./install_boost.sh && ./install_bullet.sh && ./roboschool_compile_and_graft.sh"

# install pip package and test
RUN pip install -e /code
RUN python -c "import gym; gym.make('roboschool:RoboschoolAnt-v1').reset()"
```
and it seems to build correctly.

doviettung96, 2019-05-07:
Yeah. Basically, I have both python2.7 and python3.5. The installation uses python2.7. Maybe that is the cause. I will try it. Thanks.

pzhokhov, 2019-05-07:
@doviettung96 yeah, that can definitely cause problems. In fact, the main reason why we install boost from source instead of a system package is to compile it for python3 instead of system-default 2.7. If you explicitly need roboschool to work with python 2.7, proceed at your own risk :) (i.e. it is likely possible, but we don't have it tested)

On the other note, as now we have a working docker ubuntu-based docker image (in my comment above), I think this issue can be closed safely.

closed 2019-05-07 by pzhokhov

doviettung96, 2019-05-07:
Thank you. Basically, I prefer python3. I am waiting for the docker image.

marysavari, 2019-07-16:
Hi. I am getting the following error. I am new in all of this stuff. Do you know what is the reason? Thanks.

from roboschool  import cpp_household  as cpp_household

ImportError: cannot import name 'cpp_household' from 'roboschool' (C:\Users\Mari\Miniconda3\lib\site-packages\roboschool\__init__.py)

pzhokhov, 2019-07-16:
Hi @marysavari ! In short, roboschool does not support windows :(  Some code of roboschool is written in c++ and relies on external libraries (Qt5, boost and bullet), as such, if you manage to install them on windows, it is possible to compile and link roboschool binaries on windows, but this will not be an easy road. 
If at all possible, I would recommend using linux or osx machine;  or a virtual machine / docker container with linux.

marysavari, 2019-07-17:
@pzhokhov Thank you very much for the prompt reply. Using linux would be easier for me.