openai/roboschool #178: Can't render on headless server

Chris-Carvelli, 2019-03-15:
I've installed roboschool from pip on a Ubuntu server (18.10) without an x server. I've always rendered other gym environments using xvfb and the Monitor wrapper, but I can't with any roboschool env.

my setup:
- python 3.6.7
- gym 0.12.0

test code:
```
import gym
import roboschool

env = gym.make('RoboschoolAnt-v1')
env.reset()
env.render()

```

error (same without xvfb):
```
xvfb-run -a -s "-screen 0 640x480x24" python3 test.py
Traceback (most recent call last):
  File "test.py", line 6, in <module>
    env.render()
  File "/home/chca/.local/lib/python3.6/site-packages/gym/core.py", line 244, in render
    return self.env.render(mode, **kwargs)
TypeError: render() missing 1 required positional argument: 'close'

```

olegklimov, 2019-03-15:
Oh year, right. Parameter `close` was removed from gym some time ago.

Chris-Carvelli, 2019-03-20:
Solved editing `render` in `gym_myjoco_xml_env.py` line 77:
```
    def render(self, mode, close):
        if close:
            return
        if mode=="human":
            self.scene.human_render_detected = True
            return self.scene.cpp_world.test_window()
        elif mode=="rgb_array":
            self.camera_adjust()
            rgb, _, _, _, _ = self.camera.render(False, False, False) # render_depth, render_labeling, print_timing)
            rendered_rgb = np.fromstring(rgb, dtype=np.uint8).reshape( (self.VIDEO_H,self.VIDEO_W,3) )
            return rendered_rgb
        else:
            assert(0)
```

salfityj, 2019-03-21:
Receiving the same error as original post. I'm seeing the same definition of `render()` in `gym_mujoco_xml_env.py` file, but still receiving this error. Sounds like a path error?

salfityj, 2019-03-21:
Also,  `env.render(close="True")` and `env.render(close="False")` is accepted syntax, however, I'm not seeing any output on the screen.

Chris-Carvelli, 2019-03-22:
Sorry, I copied the backup instead of my modification. Please take a look at it again (removed `close` param and deleted first two rows inside the function)

If you're rendering normally the above change is superfluous, you can just pass the `close` parameter to `render` and you should good to go. But if you're using `gym.wrapper.Monitor` to save the output to a file it won't be enough, since also `env.reset()` will trigger a rendering pass, and reset doesn't accept `close`

salfityj, 2019-03-22:
Boom that worked! Thanks.

PR #180 (2019-05-03): Remove close kwarg in render
Has been removed upstream over a year ago and can lead to crashes
because the close arg isn't passed.

Fixes #178, @Chris-Carvelli came up with the fix, also see https://github.com/openai/roboschool/issues/178#issuecomment-475508426.
closed 2019-05-03 by pzhokhov