openai/roboschool #181: ImportError: libharfbuzz.so.0: cannot open shared object file: No such file or directory

cvik1, 2019-03-30:
Hi I have been trying to train an agent on RoboSchoolAnt in an Amazon Web Services instance. 
I am using the Deep Learning AMI (Amazon Linux) Version 21.2 (ami-0a06156dfe1431263) on a p2.xlarge EC2 instance. 

I used `sudo pip install gym` and `sudo pip install roboschool`. I can import and use gym fine but when I try to `import roboschool` either in a file or in the python shell I get the following error:
`Traceback (most recent call last):
  File "train.py", line 10, in <module>
    import roboschool
  File "/home/ec2-user/anaconda3/lib/python3.6/site-packages/roboschool/__init__.py", line 112, in <module>
    from roboschool.gym_pendulums import RoboschoolInvertedPendulum
  File "/home/ec2-user/anaconda3/lib/python3.6/site-packages/roboschool/gym_pendulums.py", line 1, in <module>
    from roboschool.scene_abstract import SingleRobotEmptyScene
  File "/home/ec2-user/anaconda3/lib/python3.6/site-packages/roboschool/scene_abstract.py", line 12, in <module>
    from roboschool  import cpp_household   as cpp_household
ImportError: libharfbuzz.so.0: cannot open shared object file: No such file or directory`

I have been looking at similar issues using roboschool and tried most of what has been suggested with no success. Most of the packages I have been instructed to install give me the error : 
`Could not find a version that satisfies the requirement cpp_household (from versions: )
No matching distribution found for cpp_household`

Thanks for any suggestions you might have.

mbhenaff, 2019-06-26:
Hello, I am also encountering this issue. Any update on this? Thanks!

cvik1, 2019-06-26:
Hi, so I couldn't figure out how to fix all the dependency issues so I ended up using a library called [pybulletgym](https://github.com/benelot/pybullet-gym). This worked great and as far as I can tell there were no significant differences in the way the physics engine worked with the models using this package. Good luck!

closed 2019-06-26 by cvik1

mohit242, 2019-07-18:
Hi, I found a fix for this issue:
`apt install libharfbuzz0b`

Eletroboss, 2020-11-12:
> Hi, I found a fix for this issue:
> `apt install libharfbuzz0b`

That worked for me, thanks!