openai/roboschool #183: QT segfault on redhat with anaconda

matthieu637, 2019-04-11:
Hello,
I'm getting a segfault on `import roboschool` because of a QT problem.
```
gdb python -ex 'r -c "import roboschool"
...
(gdb) bt
#0  0x00002aaaba065600 in ?? () from /home/190977/git/pipinst/build/anaconda/bin/../lib/libQt5Core.so.5
#1  0x00002aaaba18a00d in QObject::QObject(QObjectPrivate&, QObject*) () from /home/190977/git/pipinst/build/anaconda/bin/../lib/libQt5Core.so.5
#2  0x00002aaaba106a49 in QIODevice::QIODevice(QIODevicePrivate&, QObject*) () from /home/190977/git/pipinst/build/anaconda/bin/../lib/libQt5Core.so.5
#3  0x00002aaaba0f804f in QBuffer::QBuffer(QByteArray*, QObject*) () from /home/190977/git/pipinst/build/anaconda/bin/../lib/libQt5Core.so.5
#4  0x00002aaab9184a7b in sanity_checks() () at python-binding.cpp:589
#5  0x00002aaab9184c50 in cpp_household_init() () at python-binding.cpp:607
#6  0x00002aaab9186395 in init_module_cpp_household_d() () at python-binding.cpp:724
```

I tried 3 scenarios : 
1) using pip install and QT libraries provided by Anaconda
2) build roboschool from source with QT libraries provided by the system (however as I don't have admin rights on this machine, I only extract system package locally and played with LD_LIBRARY_PATH, etc.)
3) build roboschool from source with QT libraries provided by Anaconda

In all scenarios, I end up with the same segfault.

Do you have any hint how to fix my QT installation (it's a headless computer) ? or to debug it ?
I tried to played with QT_PLUGIN_PATH as mention in sanity_checks, but it didn't change anything. 

I should mention that it is a Red Hat 6.5 (yum-3.2) cluster where qt5 is not installed but some qt3 packages are.
I tried in a similar CentOS 7.3 (yum-3.4) cluster where there isn't any qt package installed and the first scenario worked.

pzhokhov, 2019-05-03:
Hi @matthieu637! Yes, you are right, roboschool wheels are not, unfortunately, truly manylinux-compatible; so you'll need to install from source. As you specified that you tried installing from source with locally-unpacked qt libraries, I would check that the file `cpp_household.so` is indeed linked with the correct libraries (via `ldd` or a similar tool). Unfortunately, I have no experience making roboschool work on redhat, so I'll leave this issue open for a more specific input from the community.

matthieu637, 2019-05-22:
@pzhokhov Thank you for trying to help.

Indeed, I checked that the linked library were the right one.
```
(base) [190977@mu01 ~]$ ldd git/aaal/build/roboschool/roboschool/cpp_household.so
        linux-vdso.so.1 =>  (0x00007fff47bff000)
        libm.so.6 => /lib64/libm.so.6 (0x00002ad19dadb000)
        libGL.so.1 => /usr/lib64/libGL.so.1 (0x00002ad19dd5f000)
        libQt5OpenGL.so.5 => /home/190977/git/aaal/build/root/lib64/libQt5OpenGL.so.5 (0x00002ad19dfbe000)
        libQt5Widgets.so.5 => /home/190977/git/aaal/build/root/lib64/libQt5Widgets.so.5 (0x00002ad19e217000)
        libQt5Gui.so.5 => /home/190977/git/aaal/build/root/lib64/libQt5Gui.so.5 (0x00002ad19eab9000)
        libQt5Core.so.5 => /home/190977/git/aaal/build/root/lib64/libQt5Core.so.5 (0x00002ad19f221000)
        libBulletSoftBody.so.2.87 => /home/190977/git/aaal/build/roboschool/roboschool/cpp-household/bullet_local_install/lib/libBulletSoftBody.so.2.87 (0x00002ad19f902000)
        libBulletDynamics.so.2.87 => /home/190977/git/aaal/build/roboschool/roboschool/cpp-household/bullet_local_install/lib/libBulletDynamics.so.2.87 (0x00002ad19fb58000)
        libBulletCollision.so.2.87 => /home/190977/git/aaal/build/roboschool/roboschool/cpp-household/bullet_local_install/lib/libBulletCollision.so.2.87 (0x00002ad19fe31000)
        libLinearMath.so.2.87 => /home/190977/git/aaal/build/roboschool/roboschool/cpp-household/bullet_local_install/lib/libLinearMath.so.2.87 (0x00002ad1a015e000)
        libassimp.so.4 => /home/190977/git/aaal/build/root/lib/libassimp.so.4 (0x00002ad1a037f000)
        libstdc++.so.6 => /home/190977/git/aaal/build/spack/opt/spack/linux-rhel6-x86_64/gcc-4.4.7/gcc-7.3.0-x6zutiv6lvpme555ydnxobwh6e37rmqm/lib64/libstdc++.so.6 (0x00002ad1a0e04000)
        libBullet3Common.so.2.87 => /home/190977/git/aaal/build/roboschool/roboschool/cpp-household/bullet_local_install/lib/libBullet3Common.so.2.87 (0x00002ad1a11de000)
        libBulletInverseDynamics.so.2.87 => /home/190977/git/aaal/build/roboschool/roboschool/cpp-household/bullet_local_install/lib/libBulletInverseDynamics.so.2.87 (0x00002ad1a13e1000)
        libPhysicsClientC_API.so.2.87 => /home/190977/git/aaal/build/roboschool/roboschool/cpp-household/bullet_local_install/lib/libPhysicsClientC_API.so.2.87 (0x00002ad1a15fb000)
        libboost_python36.so.1.69.0 => /home/190977/git/aaal/build/root/lib/libboost_python36.so.1.69.0 (0x00002ad1a1a72000)
        libgcc_s.so.1 => /home/190977/git/aaal/build/spack/opt/spack/linux-rhel6-x86_64/gcc-4.4.7/gcc-7.3.0-x6zutiv6lvpme555ydnxobwh6e37rmqm/lib64/libgcc_s.so.1 (0x00002ad1a1cad000)
        libc.so.6 => /lib64/libc.so.6 (0x00002ad1a1ec6000)
        libglapi.so.0 => /usr/lib64/libglapi.so.0 (0x00002ad1a225a000)
        libXext.so.6 => /usr/lib64/libXext.so.6 (0x00002ad1a2480000)
        libXdamage.so.1 => /usr/lib64/libXdamage.so.1 (0x00002ad1a2693000)
        libXfixes.so.3 => /usr/lib64/libXfixes.so.3 (0x00002ad1a2895000)
        libX11-xcb.so.1 => /usr/lib64/libX11-xcb.so.1 (0x00002ad1a2a9c000)
        libX11.so.6 => /usr/lib64/libX11.so.6 (0x00002ad1a2c9d000)
        libxcb-glx.so.0 => /usr/lib64/libxcb-glx.so.0 (0x00002ad1a2fda000)
        libxcb-dri2.so.0 => /usr/lib64/libxcb-dri2.so.0 (0x00002ad1a31f1000)
        libxcb.so.1 => /usr/lib64/libxcb.so.1 (0x00002ad1a33f5000)
        libXxf86vm.so.1 => /usr/lib64/libXxf86vm.so.1 (0x00002ad1a3613000)
        libdrm.so.2 => /usr/lib64/libdrm.so.2 (0x00002ad1a3819000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00002ad1a3a24000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00002ad1a3c41000)
        libselinux.so.1 => /lib64/libselinux.so.1 (0x00002ad1a3e46000)
        libgobject-2.0.so.0 => /lib64/libgobject-2.0.so.0 (0x00002ad1a4065000)
        libgthread-2.0.so.0 => /lib64/libgthread-2.0.so.0 (0x00002ad1a42b2000)
        librt.so.1 => /lib64/librt.so.1 (0x00002ad1a44b6000)
        libglib-2.0.so.0 => /lib64/libglib-2.0.so.0 (0x00002ad1a46be000)
        libpng12.so.0 => /usr/lib64/libpng12.so.0 (0x00002ad1a49c4000)
        libz.so.1 => /lib64/libz.so.1 (0x00002ad1a4bea000)
        libicui18n.so.42 => /usr/lib64/libicui18n.so.42 (0x00002ad1a4e00000)
        libicuuc.so.42 => /usr/lib64/libicuuc.so.42 (0x00002ad1a5197000)
        libicudata.so.42 => /usr/lib64/libicudata.so.42 (0x00002ad1a54e9000)
        /lib64/ld-linux-x86-64.so.2 (0x0000003363000000)
        libgomp.so.1 => /home/190977/git/aaal/build/spack/opt/spack/linux-rhel6-x86_64/gcc-4.4.7/gcc-7.3.0-x6zutiv6lvpme555ydnxobwh6e37rmqm/lib64/libgomp.so.1 (0x00002ad1a662f000)
        libutil.so.1 => /lib64/libutil.so.1 (0x00002ad1a685f000)
        libXau.so.6 => /usr/lib64/libXau.so.6 (0x00002ad1a6a63000)
```
If I isolate the sanity_checks() function in an independent c++ file: there is no error with the exact same qt libraries.

```
#include <fstream>
#include <iostream> 
#include <QtWidgets/QApplication>
#include <QtWidgets/QDesktopWidget>
#include <QtGui/QWindow>
#include <QtCore/QElapsedTimer>
#include <QtCore/QBuffer>

void sanity_checks()
{
        float t;
        int r = sscanf("5.5", "%f", &t);
        if (r!=1 || t!=5.5f) {
                fprintf(stderr, "Sanity check failed: sscanf(\"5.5\", \"%%f\", ...)) doesn't work. Fix it by LC_ALL=C or LC_NUMERIC=en_GB.UTF-8\n");
                fprintf(stderr, "(because a lot of .xml files use \".\" as decimal separator, scanf should work on them!)\n");
[... 42 lines cut ...]
        libicuuc.so.42 => /usr/lib64/libicuuc.so.42 (0x0000003371c00000)
[... cut ...]
        libxcb-dri2.so.0 => /usr/lib64/libxcb-dri2.so.0 (0x0000003f00c00000)
        libxcb.so.1 => /usr/lib64/libxcb.so.1 (0x0000003366400000)
        libXxf86vm.so.1 => /usr/lib64/libXxf86vm.so.1 (0x0000003eff800000)
        libdrm.so.2 => /usr/lib64/libdrm.so.2 (0x0000003f01000000)
        libselinux.so.1 => /lib64/libselinux.so.1 (0x0000003364c00000)
        libXau.so.6 => /usr/lib64/libXau.so.6 (0x0000003366800000)
QT_PLUGIN_PATH=~/git/aaal/build/root/lib64/qt5/plugins/ ./a.out
#return 0
```