openai/roboschool #184: Unable to open file ".../meshes_unplug"

ishaanjain, 2019-04-16:
I'm trying to run a simple gym program using the RoboschoolAtlasForwardWalk-v1 environment. Here's my code:
```
import gym
import roboschool

env = gym.make('RoboschoolAtlasForwardWalk-v1')
for i_episode in range(20):
    observation = env.reset()
    for t in range(100):
        env.render()
        action = env.action_space.sample()
        observation, reward, done, info = env.step(action)
        if done:
            print("Episode finished after {} timesteps".format(t+1))
            break
env.close()
```
But I'm getting the following error:
```
cannot load '/Users/ishaanjain/anaconda3/envs/atlas-walk/lib/python3.6/site-packages/roboschool/models_robot/atlas_description/meshes_unplug': Unable to open file "/Users/ishaanjain/anaconda3/envs/atlas-walk/lib/python3.6/site-packages/roboschool/models_robot/atlas_description/meshes_unplug".
```
Both the gym and roboschool packages were installed into an Anaconda package environment, and there doesn't seem to be any problem with importing the packages themselves. I've inspected the relevant directory where the issue is (`/atlas_description`) and I found that it contains a folder called `meshes_unplugged` but no `meshes_unplug` file or folder. I've played around with my setup and browsed the roboschool issues list for a solution, but haven't found one. Any idea on whether this is a bug with the roboschool library or more likely an issue on my end?

OS is MacOS.

muupan, 2019-05-08:
I'm facing the same issue. It seems the path is limited to 128 chars? I'm using macOS Mojave and I installed `roboschool==1.0.46` from pypi.

keiohta, 2019-06-10:
I got an exactly the same error with Anaconda, Mac OS Mojave 10.14.5, and `roboschool==1.0.48` installed from pypi.
```bash
# Reproduce the result
$ python
Python 3.7.3 (default, Mar 27 2019, 16:54:48) 
[Clang 4.0.1 (tags/RELEASE_401/final)] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import gym, roboschool
>>> env = gym.make("RoboschoolAtlasForwardWalk-v1")
>>> env.reset()
array([-5.0000000e-01,  2.1938172e-07,  1.0000000e+00,  0.0000000e+00,
        0.0000000e+00,  0.0000000e+00,  0.0000000e+00, -0.0000000e+00,
       -1.0129737e-01,  0.0000000e+00, -2.7300680e-01,  0.0000000e+00,
       -2.2963487e-02,  0.0000000e+00,  3.8692439e-01,  0.0000000e+00,
        2.7873237e-02,  0.0000000e+00, -9.6822846e-01,  0.0000000e+00,
       -9.2063069e-01,  0.0000000e+00, -1.0073267e+00,  0.0000000e+00,
        2.6331929e-02,  0.0000000e+00, -5.0000000e+00,  0.0000000e+00,
       -2.5171056e-01,  0.0000000e+00, -3.0473593e-01,  0.0000000e+00,
       -5.6794822e-02,  0.0000000e+00, -1.0301926e+00,  0.0000000e+00,
        1.0308630e+00,  0.0000000e+00, -1.0177650e+00,  0.0000000e+00,
       -2.0845551e-02,  0.0000000e+00,  5.0000000e+00,  0.0000000e+00,
       -6.0120189e-01,  0.0000000e+00,  4.7180455e-02,  0.0000000e+00,
        4.6646124e-01,  0.0000000e+00, -1.0220735e+00,  0.0000000e+00,
        2.9406804e-01,  0.0000000e+00, -9.1966977e-03,  0.0000000e+00,
        4.9638817e-01,  0.0000000e+00,  2.7587848e-02,  0.0000000e+00,
        4.2250580e-01,  0.0000000e+00, -9.1844368e-01,  0.0000000e+00,
        1.5770590e-01,  0.0000000e+00, -6.2665530e-02,  0.0000000e+00,
        0.0000000e+00,  0.0000000e+00], dtype=float32)
>>> env.render()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/keiohta/anaconda3/envs/tf2/lib/python3.7/site-packages/gym/core.py", line 230, in render
    return self.env.render(mode, **kwargs)
  File "/Users/keiohta/anaconda3/envs/tf2/lib/python3.7/site-packages/roboschool/gym_urdf_robot_env.py", line 90, in render
    return self.scene.cpp_world.test_window()
RuntimeError: cannot load '/Users/keiohta/anaconda3/envs/tf2/lib/python3.7/site-packages/roboschool/models_robot/atlas_description/meshes_unplugged/pelvis': Unable to open file "/Users/keiohta/anaconda3/envs/tf2/lib/python3.7/site-packages/roboschool/models_robot/atlas_description/meshes_unplugged/pelvis".

# Check if `pelvis` exists
$ ls /Users/keiohta/anaconda3/envs/tf2/lib/python3.7/site-packages/roboschool/models_robot/atlas_description/meshes_unplugged/
l_foot.dae	l_lleg.dae	l_uglut.dae	ltorso.dae	pelvis.dae	r_foot.dae	r_lleg.dae	r_talus.dae	r_uglut.dae	utorso.dae
l_lglut.dae	l_talus.dae	l_uleg.dae	mtorso.dae	r_clav.dae	r_lglut.dae	r_scap.dae	r_uarm.dae	r_uleg.dae
```

Irunmiles, 2019-07-08:
> I'm facing the same issue. It seems the path is limited to 128 chars? I'm using macOS Mojave and I installed `roboschool==1.0.46` from pypi.

**macOS Mojave 10.14.5
conda 4.6.4**
I faced the same problem. The path is limited to 127 chars. And I know how to fix it.

### Step 1
`Conda install --target=/your/path roboschool` 
You need to install roboschool package into a different directory. It allows you to reduce the path's length. 
P.S. more details are [here](https://stackoverflow.com/a/19404371) and I used this command
`Conda install --target=/Users/khakimov/Anaconda/roboschool roboschool`
### Step 2
In my case command `pip show roboschool` shows me nothing. We need to modify **sys.path** and help _pip_ to see this new directory.
#### fast way
```
import sys
sys.path.append('/your/path')
import roboschool
import gym
```
You need to add the first 2 lines of code each time before import roboschool. **Always!**
P.S. more details are [here](https://stackoverflow.com/a/38485710) and I used this
```
import sys
sys.path.append('/Users/khakimov/Anaconda/roboschool')
```
#### long way
Find the _site.py_ somewhere in your _Anaconda/envs/env/python_ directory. In my case I have found it here: _/Users/khakimov/Anaconda/anaconda3/envs/atlas/lib/python3.5/site.py_. Open it and read comments. It gives your intuition what is going on.
Next let's jump one more folder deeper into the _site-package_ and create there the _roboschool.pth_ file. 
_/Users/khakimov/Anaconda/anaconda3/envs/atlas/lib/python3.5/site-packages/roboschool.pth_
Open this file with any text reader and write the path where you installed roboschool package (` /your/path `from **Step 1**), in my case I wrote `/Users/khakimov/Anaconda/roboschool`
Now in your terminal you can use command `pip show roboschool`. You will see something like that:
```
Name: roboschool
Version: 1.0.48
Summary: OpenAI Household Simulator: mobile manipulation using Bullet
Home-page: https://github.com/openai/roboschool
Author: None
Author-email: None
License: UNKNOWN
Location: /Users/khakimov/Anaconda/roboschool
Requires: gym
Required-by:
```

P.S. more details are [here](https://stackoverflow.com/a/38485710)