openai/roboschool #186: Can someone please tell me where can I find  information on maximum reward  for roboschool environments?

elitalobo, 2019-04-20:


elitalobo, 2019-04-20:
Unable to find any documentation containing the baseline score or maximum reward details for roboschool environments.

wpumacay, 2019-04-21:
Hi @elitalobo 

I'm not sure if this is helpful, but you could take a look at the [gym_forward_walker.py](https://github.com/openai/roboschool/blob/0504ac029c858b72c862372d269ed7564d591d9f/roboschool/gym_forward_walker.py#L119) file. Inside the step method (line 119) the reward is computed from various components that make the whole reward function (+1 per alive-step, progress, actuation cost, ...). You could take a look at how they compute the rewards and get an idea of the actual ranges (I think the reward functions are not documented).

Perhaps what you are looking for is a standard range in which the rewards are going to be, like in the environments from dm_control. In their [paper](https://arxiv.org/pdf/1801.00690.pdf) they specify that the rewards belong to a range of [0,1], and in their code they make the required scaling to make sure these ranges are in that range. For example, in the [walker.py](https://github.com/deepmind/dm_control/blob/master/dm_control/suite/walker.py#L143), they scale the components of the reward function to obtain a standard range of [0,1].