openai/roboschool #187: AttributeError: 'NoneType' object has no attribute 'human_render_detected'

toandx, 2019-05-03:
I install roboschool by pip in ubuntu 16.04 64 bit and python 3.5.2 and get this error
>>> import roboschool
>>> import gym
>>> env = gym.make('RoboschoolAnt-v1')
>>> env.render()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/gym/core.py", line 237, in render
    return self.env.render(mode, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/roboschool/gym_mujoco_xml_env.py", line 81, in render
    self.scene.human_render_detected = True
AttributeError: 'NoneType' object has no attribute 'human_render_detected'
Can anyone help me! Please!

pzhokhov, 2019-05-03:
the error message is admittedly not very telling, but you need to call `env.reset()` before `env.render()`, so the whole snippet should look like:
```python
import gym
env = gym.make('RoboschoolAnt-v1')
env.reset() .  # will throw errors without this line
env.render()
```

toandx, 2019-05-04:
But when I run env.reset(), always "Illegal instruction (core dumped)". I have tried install the lastest gym (0.12.1) and reinstall roboschool by pip but it not work.

Python 3.5.2 (default, Nov 12 2018, 13:43:14) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gym
>>> import roboschool
>>> gym.__version__
'0.12.1'
>>> env=gym.make('RoboschoolAnt-v1')
>>> env.reset()
Illegal instruction (core dumped)

pzhokhov, 2019-05-04:
huh... okay, that looks like some library is missing / of the wrong version. What I tried to do to reproduce was this - in ubuntu:16.04 docker container I ran
`apt-get -y update && apt-get install -y python3-dev python3-pip libgl1-mesa-dev libharfbuzz-dev libpcre16-3 && pip3 install roboschool`
and then your snippet, and it worked without any errors (returned first Ant observation). 
What that tells me is that on a vanilla version of ubuntu 16.04 things should work. I would try the following:

1. `apt-get -y update && apt-get upgrade ibgl1-mesa-dev libharfbuzz-dev libpcre16-3`, then reinstall roboschool (`pip3 uninstall -y roboschool && pip3 install roboschool`), and try again

2.  If that does not help, find the installation folder for roboschool (via, e.g. `python3 -c "import roboschool; print(roboschool.__file__)`), go to that folder, and run `ldd cpp_household.so | sort | grep -v ".lib" - this will list all libraries roboschool compiled C++ code depends on, excluding the ones shipped with the wheel. In my docker image, this results in: 
```
root@ba31a4266ef1:/usr/local/lib/python3.5/dist-packages/roboschool# ldd cpp_household.so  | sort | grep -v '.libs'
	/lib64/ld-linux-x86-64.so.2 (0x00007f0d4104a000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1 (0x00007f0d40822000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f0d3b8b5000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f0d3b57b000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f0d38a2c000)
	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007f0d3bcbd000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f0d38826000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f0d3bec0000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f0d3bab7000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007f0d3ad35000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f0d3cf56000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0d3a702000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007f0d3ab23000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f0d3cb13000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f0d3857c000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f0d3d320000)
	libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007f0d3c0d2000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f0d39247000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f0d38356000)
	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f0d3a4a4000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f0d40a96000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f0d36669000)
	libpcre16.so.3 => /usr/lib/x86_64-linux-gnu/libpcre16.so.3 (0x00007f0d39769000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f0d36444000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f0d3a906000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f0d3903f000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f0d38c30000)
	libxcb-dri2.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0 (0x00007f0d3b15d000)
	libxcb-dri3.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 (0x00007f0d3c910000)
	libxcb-glx.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0 (0x00007f0d3b362000)
	libxcb-present.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-present.so.0 (0x00007f0d3c70d000)
	libxcb-sync.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007f0d3c506000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f0d3af3b000)
	libxshmfence.so.1 => /usr/lib/x86_64-linux-gnu/libxshmfence.so.1 (0x00007f0d3c303000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f0d3cd3c000)
	linux-vdso.so.1 =>  (0x00007fff0adbf000)
```
If there are some mismatches, I would try to fix those (by installing a correct version of a library), and then reinstall roboschool again.

3. Install roboschool from source

toandx, 2019-05-04:
In step 2, I miss file libstdc++.so.6 when run **ldd cpp_household.so | sort | grep -v '.libs'**. But when I try sudo apt-get install libstdc++6, it print **"libstdc++6 is already the newest version (5.4.0-6ubuntu1~16.04.11).**". How can I fix this bug?

closed 2019-05-04 by toandx

tamirblum, 2019-10-01:
I am getting a similar issue to toandx with the core dumped illegal instruction. Did anyone identify the cause?