openai/roboschool #191: AttributeError: 'EnvSpec' object has no attribute 'timestep_limit'

djbyrne, 2019-05-25:
I have just tried to run Roboschool. It installs fine but I am unable to carry out the env.step(action) function, receiving the following error

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-108-687456da6d84> in <module>
      1 # Populate replay buffer
----> 2 observe(env, replay_buffer, OBSERVATION)

<ipython-input-102-2864dc5e5e38> in observe(env, replay_buffer, observation_steps)
     15     while time_steps < observation_steps:
     16         action = env.action_space.sample()
---> 17         new_obs, reward, done, _ = env.step(action)
     18 
     19         replay_buffer.add((obs, new_obs, action, reward, done))

~/Documents/Workspace/RL/Repos/gym/gym/wrappers/time_limit.py in step(self, action)
     13     def step(self, action):
     14         assert self._elapsed_steps is not None, "Cannot call env.step() before calling reset()"
---> 15         observation, reward, done, info = self.env.step(action)
     16         self._elapsed_steps += 1
     17         if self._elapsed_steps >= self._max_episode_steps:

~/anaconda3/envs/fastai/lib/python3.6/site-packages/roboschool/gym_forward_walker.py in step(self, a)
    126 
    127         self.frame  += 1
--> 128         if (done and not self.done) or self.frame==self.spec.timestep_limit:
    129             self.episode_over(self.frame)
    130         self.done   += done   # 2 == 1+True

AttributeError: 'EnvSpec' object has no attribute 'timestep_limit'

```

I am running on Ubuntu 18.04 with python 3.6.8

I have tried installing from pip and from source with no luck.

Any help would be appreciated

AroMorin, 2019-05-29:
I can confirm that this also happens with me. The pip version of roboschool is 1.0.46. In line 128 of gym_forward_walker it does call self.spec.timestep_limit. Looking at current gym_forward_walker in source file, the line calls self.spec.max_episode_steps instead.

I am using python 3.6.7, pip 19.1.1 and Ubuntu 18.04.

djbyrne, 2019-05-29:
I managed to get it working today by using the exact same requirements.txt file from my osx environment.
Once I install these libraries with requirements it worked fine.

'''
bsl-py==0.7.1
appdirs==1.4.3
appnope==0.1.0
asn1crypto==0.24.0
astor==0.7.1
astroid==2.2.5
atari-py==0.1.7
atomicwrites==1.3.0
attrs==19.1.0
autobahn==19.3.3
backcall==0.1.0
baselines==0.1.5
black==19.3b0
bleach==1.5.0
box2d-py==2.3.8
certifi==2019.3.9
cffi==1.12.2
chardet==3.0.4
Click==7.0
cloudpickle==0.8.1
cmarkgfm==0.4.2
coverage==4.5.3
coveralls==1.6.0
cryptography==2.6.1
cycler==0.10.0
Cython==0.29.6
decorator==4.4.0
defusedxml==0.5.0
dill==0.2.9
docker-py==1.10.3
docker-pycreds==0.2.1
docopt==0.6.2
docutils==0.14
entrypoints==0.3
fastzbarlight==0.0.14
future==0.17.1
gast==0.2.2
glfw==1.7.1
grpcio==1.11.0
-e git+https://github.com/openai/gym.git@c1285d52555cf611f91cefcca7654a12cd32ff3b#egg=gym
h5py==2.9.0
html5lib==0.9999999
idna==2.8
imageio==2.5.0
ipykernel==5.1.0
ipython==7.4.0
ipython-genutils==0.2.0
ipywidgets==7.4.2
isort==4.3.16
jedi==0.13.3
Jinja2==2.10
joblib==0.13.2
jsonschema==3.0.1
jupyter==1.0.0
jupyter-client==5.2.4
jupyter-console==6.0.0
jupyter-core==4.4.0
Keras==2.2.4
Keras-Applications==1.0.7
Keras-Preprocessing==1.0.9
kiwisolver==1.0.1
lazy-object-proxy==1.3.1
lockfile==0.12.2
Markdown==3.1
MarkupSafe==1.1.1
matplotlib==3.0.3
mccabe==0.6.1
mock==2.0.0
more-itertools==7.0.0
mpi4py==3.0.1
nbconvert==5.4.1
nbformat==4.4.0
notebook==5.7.8
numpy==1.14.5
olefile==0.46
opencv-python==4.0.0.21
pandas==0.24.2
pandocfilters==1.4.2
parso==0.3.4
pbr==5.1.3
pexpect==4.6.0
pickleshare==0.7.5
Pillow==5.4.1
pkginfo==1.5.0.1
pluggy==0.9.0
progressbar2==3.39.3
prometheus-client==0.6.0
prompt-toolkit==2.0.9
protobuf==3.6.1
psutil==5.6.1
ptan==0.3
ptyprocess==0.6.0
py==1.8.0
pybullet==2.4.8
pycparser==2.19
pyglet==1.3.2
Pygments==2.3.1
pylint==2.3.1
PyOpenGL==3.1.0
pyOpenSSL==19.0.0
pyparsing==2.3.1
pyrsistent==0.14.11
PySocks==1.6.8
pytest==3.10.1
pytest-cov==2.6.1
python-dateutil==2.8.0
python-utils==2.3.0
pytz==2018.9
PyYAML==5.1
pyzmq==18.0.0
qtconsole==4.4.3
readme-renderer==24.0
requests==2.10.0
requests-toolbelt==0.9.1
roboschool==1.0.46
scipy==1.2.1
seaborn==0.8.1
Send2Trash==1.5.0
six==1.12.0
-e git+https://github.com/openai/spinningup.git@97c8c342c45e5bb51005a8515df23ba9c48f0782#egg=spinup
tensorboard==1.7.0
tensorboard-pytorch==0.7.1
tensorflow==1.7.1
tensorflow-estimator==1.13.0
termcolor==1.1.0
terminado==0.8.1
testpath==0.4.2
toml==0.10.0
torch==1.0.1.post2
torchvision==0.2.2
tornado==6.0.2
tqdm==4.31.1
traitlets==4.3.2
twine==1.13.0
txaio==18.8.1
typed-ast==1.3.1
urllib3==1.24.1
wcwidth==0.1.7
webencodings==0.5.1
websocket-client==0.56.0
Werkzeug==0.15.2
widgetsnbextension==3.4.2
wrapt==1.11.1
zmq==0.0.0
'''
I imagine it has to do with the gym version, which I had installed from source in this environment. Hope this helps

AroMorin, 2019-05-29:
I personally didn't install gym from source, just using pip. And it's on the latest version 0.12.5.

I don't really understand what you did, but if it worked, then great! I'll hope someone will have a better fix than this?!

bhatiaabhinav, 2019-05-30:
gym==0.12.5 was released 6 days ago. It broke since then. Roboschool==1.0.46 works fine with gym==0.12.1.

Thus the following solves the issue:
```
pip install roboschool==1.0.46
pip install gym==0.12.1
```

sibyjackgrove, 2019-06-11:
I ran into the same issue as well on Windows OS. But downgrading to gym==0.12.1 worked.

rickstaa, 2020-07-23:
For future reference, to make it compatible with newer version of gym, you can also change the line to:

```python
MAX_STEPS = env._max_episode_steps 
```

TachikakaMin, 2021-02-19:
change timestep_limit to max_episode_steps

https://github.com/tensorflow/agents/commit/96e7b31d71023217ffe9e7febff2ef2c5db88513