openai/roboschool #193: cpp_household: libpcre16.so.3 => not found

SolbiatiAlessandro, 2019-06-09:
I am trying to install roboschool on a Google AI Platform instance.

I get the following error  
`
from roboschool import cpp_household as cpp_household ImportError: libpcre16.so.3: cannot open shared object file: No such file or directory
`
Unfortunately I can't ssh into the machine and can not give more detail on how to repro. I investigated a bit and running the command `ldd cpp_household.so` I see how the machine can not find `libpcre16.so.3`. Here is the output of the command:

```
I  master-replica-0 	libicui18n.so.57 => /root/.local/lib/python3.5/site-packages/roboschool/.libs/libicui18n.so.57 (0x00007f91d39a2000) master-replica-0 
I  master-replica-0 	libicuuc.so.57 => /root/.local/lib/python3.5/site-packages/roboschool/.libs/libicuuc.so.57 (0x00007f91d35d9000) master-replica-0 
I  master-replica-0 	libpcre16.so.3 => not found master-replica-0 
I  master-replica-0 	libdouble-conversion.so.1 => /root/.local/lib/python3.5/site-packages/roboschool/.libs/libdouble-conversion.so.1 (0x00007f91d33c8000) master-replica-0 
I  master-replica-0 	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f91d30b7000) master-replica-0 
```

I tried the following:
```apt-get install libpcre-3```
And it tells me that `libpcre3 is already the newest version (2:8.38-3.1).`

Any idea on why I don't have this specific library? Any idea on how to fix it?

SolbiatiAlessandro, 2019-06-09:
I tried
```apt-get install libpcre16-3```
And I get `E: Unable to locate package libpcre16-3`

SolbiatiAlessandro, 2019-06-09:
To give more context on the missing library: [libpcre3](https://packages.debian.org/en/source/stretch/pcre3)

SolbiatiAlessandro, 2019-06-22:
any idea on this?

ameenba, 2019-07-01:
same troubles here - did you solve it?

SolbiatiAlessandro, 2019-07-01:
Hi, I tried for two weeks but without success. I even opened an issue with my Google Cloud Support but they didn't reply. I am still trying, post here if you find a solution or some hint.

CommanderCero, 2019-07-04:
So I installed roboschool just by using pip install roboschool.
Then I got the same error and I fixed it with apt-get install libpcre3**-dev**, well partially...

SolbiatiAlessandro, 2019-07-10:
@CommanderCero were you on google cloud? because it doesn't work for me

CommanderCero, 2019-07-10:
I've tested this on an Ubuntu 16 and 18 VM, aswell on an "actual" Ubuntu 18 OS and in all versions when i run pip install roboschool and afterwards try to import roboschool it throws _ImportError: libpcre16.so.3: cannot open shared object file: No such file or directory_

I've also managed to fix this issue in all version by installing libpcre3-**dev**. Well if it would really help solve the issue for other people is still questionable, as it didn't work afterwards as mentioned

leoi137, 2019-07-19:
> @CommanderCero were you on google cloud? because it doesn't work for me

I tried sudo apt-get install libpcre16-3 and it worked for me. Did you install the graphics library libgl1-mesa-dev? If you haven't I would try it using: sudo apt-get install libgl1-mesa-dev, I am not entirely sure if this will fix the problem as libpcre is a regular expressions engine.

zhan0903, 2019-08-04:
> So I installed roboschool just by using pip install roboschool.
> Then I got the same error and I fixed it with apt-get install libpcre3-**dev**, well [partially.](https://github.com/openai/roboschool/issues/195)..

What if I don't have the root right to install libpcre3-dev, can I install it by conda or other methods?

bibbygoodwin, 2019-08-04:
I'm also finding myself in a permission issue here - I'd been relying on a `pip install roboschool` working smoothly on a server I don't have sudo rights on. Are there alternative routes to installing these missing packages, or workarounds? I think the conda installation instructions [here](https://github.com/openai/roboschool/issues/122) still suppose sudo access for library installations.

realjohnward, 2019-08-24:
> I tried sudo apt-get install libpcre16-3 and it worked for me.
@leoi137 's solution worked for me.

andrei-radulescu-banu, 2021-01-12:
It's still not installing properly. On Ubuntu 18.04.5, after installing libpcre16-3:
```
$ sudo apt-get install libpcre16-3
$ python -c "import roboschool"
Illegal instruction
```
On Centos 7, libpcre16.so.3 is not available as a package.

Does not work to install roboschool from sources either. That module is deprecated, it has been replaced by PyBullet.

On MacOS, get the following error:
```
$ ./bin/setup
Error: caskroom/cask was moved. Tap homebrew/cask instead.
```

I think it would help if we knew the exact Ubuntu version this is expected to work with. Even if it's an older Ubuntu, it could be run in a Docker container.

chaennlee, 2023-07-18:
you did it? I cant solve that problem still..