openai/roboschool #197: unknown option 'qt' when running make -j4

mansur007, 2019-07-13:
when executing make -j4 following error appears:
...
-c render-simple.cpp  -o.build-release/render-simple.o -MMD -MF .build-release/render-simple.o.dep
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp bullet` -I/usr/local/include/bullet -c render-simple-primitives.cpp  -o.build-release/render-simple-primitives.o -MMD -MF .build-release/render-simple-primitives.o.dep
moc -qt=5 -o .generated/test-tool-qt4.moc test-tool-qt4.cpp
Unknown option 'qt'.
Makefile:125: recipe for target '.generated/test-tool-qt4.moc' failed
make: *** [.generated/test-tool-qt4.moc] Error 1
make: *** Waiting for unfinished jobs....

Please help, couldn't find solution with google.
Ubuntu 16.04.6, installing with conda, tried both python 3.7 and 3.6.
Thanks

apple1987, 2019-07-15:
met the same question,    gooooooooooooooooooooooooo for help~

mansur007, 2019-07-17:
> met the same question, gooooooooooooooooooooooooo for help~

sorry, I didn't get what you mean. What is gooo...?

apple1987, 2019-07-17:
Have you solved this problem?

mansur007, 2019-07-21:
> Have you solved this problem?

Unfortunately not

TheCrazyT, 2019-07-21:
just a guess, but maybe try to update pyqt?

TylerJamesMalloy, 2019-08-13:
This seems to be an issue with the line 
`
MOC  =moc -qt=5
`

I am using fedora so it may be related to a fedora specific issue, but I fixed it by replacing that line with: 
`
MOC  =moc
`

mansur007, 2019-08-24:
> This seems to be an issue with the line
> `MOC =moc -qt=5`
> 
> I am using fedora so it may be related to a fedora specific issue, but I fixed it by replacing that line with:
> `MOC =moc`

I tried that and got:
many warnings
...
...
collect2: error: ld returned 1 exit status
Makefile:133: recipe for target '../robot-test-tool_d' failed
make: *** [../robot-test-tool_d] Error 1

mansur007, 2019-08-24:
> just a guess, but maybe try to update pyqt?

doesn't work

louis-shao, 2019-09-08:
Met the same problem on ubuntu 18.04. Removing "-qt=5" works totally fine for me.