openai/roboschool #207: ImportError: cannot import name 'cpp_household' [Google Colab]

alec-tschantz, 2019-10-05:
Using `!pip install roboschool` and then `import roboschool` on a Google Colab .ipynb throws:

` /usr/local/lib/python3.6/dist-packages/roboschool/__init__.py in <module>() 
--> 112 from roboschool.gym_pendulums import RoboschoolInvertedPendulum
`

`/usr/local/lib/python3.6/dist-packages/roboschool/gym_pendulums.py in <module>()
----> 1 from roboschool.scene_abstract import SingleRobotEmptyScene`

/`usr/local/lib/python3.6/dist-packages/roboschool/scene_abstract.py in <module>()
---> 12 from roboschool  import cpp_household   as cpp_household
ImportError: cannot import name cpp_household`

This has only started occurring today - the same notebooks ran without errors yesterday. I have tried installing a previous version of roboschool (1.0.48), as (1.0.49) was committed yesterday, but the same import error is thrown. 

I notice a couple of related issues were raised recently (#203 & #202) but raising a separate issue due to the ease of reproducing on Colab.

alec-tschantz, 2019-10-05:
After resetting runtimes, the full error import error is:

`ImportError: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /usr/local/lib/python3.6/dist-packages/roboschool/.libs/libQt5Core.so.5)`

On Google colab, can confirm GLIBC 2.27 is installed (`!ldd --version -> (Ubuntu GLIBC 2.27-3ubuntu1) 2.27`

[Stack Overflow](https://unix.stackexchange.com/questions/176489/how-to-update-glibc-to-2-14-in-centos-6-5) has some suggestions for upgrading GLIBC, but it is non-trivial on Colab. 

Any suggested workarounds?

Trinkle23897, 2019-10-08:
You can try to install version 1.0.48.

pdeubel, 2019-10-09:
I had the exact same issue as you with version `1.0.49`, but it went away when I installed `1.0.48`. Maybe you can try that again, perhaps it did not refresh your version of roboschool or something like that.

AkhilSinghRana, 2019-10-12:
> I had the exact same issue as you with version `1.0.49`, but it went away when I installed `1.0.48`. Maybe you can try that again, perhaps it did not refresh your version of roboschool or something like that.

Perfect

alec-tschantz, 2019-10-15:
> I had the exact same issue as you with version `1.0.49`, but it went away when I installed `1.0.48`. Maybe you can try that again, perhaps it did not refresh your version of roboschool or something like that.

Great - `!pip install roboschool==1.0.48` does indeed work. Restarting the kernel doesn't clear installed packages, which makes sense in hindsight. Thank you very much.

closed 2019-10-15 by alec-tschantz

lhxxxxx, 2023-11-24:
but how can I install roboschool 1.0.48 in pycharm