openai/roboschool #208: unexpected keyword argument 'tags'

ZiningYin, 2020-02-10:
File "/usr/local/lib/python3.6/dist-packages/roboschool/__init__.py", line 16, in <module>
    tags={ "pg_complexity": 1*1000000 },
  File "/usr/local/lib/python3.6/dist-packages/gym/envs/registration.py", line 139, in register
    return registry.register(id, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/gym/envs/registration.py", line 133, in register
    self.env_specs[id] = EnvSpec(id, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'tags'

thavidu, 2020-02-17:
+1 I get this issue too, when trying to just import roboschool

thavidu, 2020-02-17:
Looks like this was introduced in https://github.com/openai/gym/commit/a99e8d153e94f532e5728b01b02b42e076334bf7#diff-3dff15d44236dd4f0f823a6ddb7e8c9b this change to openAI gym

thavidu, 2020-02-18:
which makes me think reverting to version 0.15.4 of gym should fix it (but i have other issues in my environment so can't confirm whether that is enough or not)

LanxinL, 2020-03-30:
Thx,  `pip install gym==0.15.4` can fix it.
Btw,  roboschool should be reverted to 1.0.48 (`pip install roboschool==1.0.48`). I get  `ImportError: /lib/x86_64-linux-gnu/libc.so.6: version 'GLIBC_2.25' not found (required by /home/llx/anaconda3/envs/openAiBaseline/lib/python3.6/site-packages/roboschool/.libs/libQt5Core.so.5)` after  reverting gym to version 0.15.4

> which makes me think reverting to version 0.15.4 of gym should fix it (but i have other issues in my environment so can't confirm whether that is enough or not)

LanxinL, 2020-03-30:
> which makes me think reverting to version 0.15.4 of gym should fix it (but i have other issues in my environment so can't confirm whether that is enough or not)

@thavidu Thx,  `pip install gym==0.15.4` can fix it.
Btw,  roboschool should be reverted to 1.0.48 (`pip install roboschool==1.0.48`). I get  `ImportError: /lib/x86_64-linux-gnu/libc.so.6: version 'GLIBC_2.25' not found (required by /home/llx/anaconda3/envs/openAiBaseline/lib/python3.6/site-packages/roboschool/.libs/libQt5Core.so.5)` after  reverting gym to version 0.15.4

Mark-bit520, 2021-07-04:
why should I revert gym to 0.15.4, why not roboschool update its version? I also install stable_baselines3, it requires gym version＞0.17.0.

sgillen, 2021-07-20:
@Mark-bit520, From the roboschool README: "Status: Archive (code is provided as-is, no updates expected)" and also: "DEPRECATED: Please use PyBullet instead". So if you want to use Roboschool you will need to downgrade your gym etc.