openai/roboschool #209: Package 'libav-tools' has no installation candidate

taniyahasija1997, 2020-03-16:
(venv) cu@cu-PowerEdge-T20:~/Tacotron-2-master$ sudo docker build -t tacotron-2_image docker/
Sending build context to Docker daemon  2.048kB
Step 1/10 : FROM continuumio/anaconda3:latest
 ---> bdb4a7e92a49
Step 2/10 : FROM tensorflow/tensorflow:latest-gpu-py3
 ---> e2a4af785bdb
Step 3/10 : RUN apt-get update
 ---> Using cache
 ---> 122e61610906
Step 4/10 : RUN apt-get install -y libasound-dev portaudio19-dev libportaudio2 libportaudiocpp0 ffmpeg libav-tools wget git vim
 ---> Running in 76ef158ee8f4
Reading package lists...
Building dependency tree...
Reading state information...
Package libav-tools is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  ffmpeg

E: Package 'libav-tools' has no installation candidate
The command '/bin/bash -c apt-get install -y libasound-dev portaudio19-dev libportaudio2 libportaudiocpp0 ffmpeg libav-tools wget git vim' returned a non-zero code: 100

#can anyone help me

mimoralea, 2020-06-20:
Check out: https://github.com/openai/roboschool/issues/26

You may need to modify the Dockerfile and replace `libav-tools` with `ffmpeg`.

AyanHBK123, 2020-12-23:
I tried to do this way but still gets the error

Ri-an, 2021-03-06:
https://github.com/Rayhane-mamah/Tacotron-2/commit/7fa13872398ae5a52c4138752416950258f5ae04

try this.

AndreiCherniaev, 2022-11-24:
> Check out: #26
> 
> You may need to modify the Dockerfile and replace `libav-tools` with `ffmpeg`.

The modern way is to use [ffmpeg](https://trac.ffmpeg.org/wiki/Capture/Webcam#dshow):

sudo apt install v4l-utils

v4l2-ctl --list-devices

ffmpeg -f video4linux2 -i /dev/video0 -vframes 1 -video_size 640x480 test.jpeg