openai/roboschool #210: import roboschool after pip install gives TypeError due to 'tags'

jithendaraa, 2020-03-29:
Can you point as to why this is happening and how it can be fixed?
pip install roboschool

import roboschool -> error

File "<stdin>", line 1, in <module>
  File "/home/jith/.local/lib/python3.7/site-packages/roboschool/__init__.py", line 16, in <module>
    tags={ "pg_complexity": 1*1000000 },
  File "/home/jith/anaconda3/lib/python3.7/site-packages/gym/envs/registration.py", line 139, in register
    return registry.register(id, **kwargs)
  File "/home/jith/anaconda3/lib/python3.7/site-packages/gym/envs/registration.py", line 133, in register
    self.env_specs[id] = EnvSpec(id, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'tags'

fearofcode, 2020-04-12:
I was able to get an install that can at least `import roboschool` without error on Ubuntu by installing these specific versions:

```
pip install roboschool==1.0.48 gym==0.15.4
```

The `gym` version comes from this comment: https://github.com/openai/roboschool/issues/208#issuecomment-587220664 .

tyler124, 2020-05-17:
`pip install roboschool==1.0.48 gym==0.15.4`

and

'import roboschool'

Results in a:
'Illegal instruction (core dumped)' 

for me

LBWNB10086, 2023-03-17:
pip install roboschool==1.0.48
ERROR:No matching distribution found for roboschool==1.0.48

TheCrazyT, 2023-03-17:
could be similar problem that is mentioned here:
https://tech.amikelive.com/node-887/how-to-resolve-error-illegal-instruction-core-dumped-when-running-import-tensorflow-in-a-python-program/

so maybe unsupported cpu for that version?