openai/roboschool #212: ImportError: libQt5OpenGL.so.5: cannot open shared object file: No such file or directory

joshua-matt, 2021-07-07:
When I try to run `python test_manual.py`, I get the above error message, which is thrown in importing `cpp_household_d` on line 11 of `scene_abstract.py`. I see that the sought file is present in the `.libs` folder. I'm using Ubuntu 20.04; is there some package I need to install?

TheCrazyT, 2021-07-08:
could be another missing dependency.
I would do:
`ldd libQt5OpenGL.so.5`
Inside the .libs folder to see if it shows any missing files.

joshua-matt, 2021-07-08:
Here is the output. What do I do with this information?
[image]

TheCrazyT, 2021-07-09:
Looks ok, else it won't show the paths.
Could you do the same for cpp_household_d.so please?

joshua-matt, 2021-07-09:
Adding the `.libs` folder to LD_LIBRARY_PATH fixed the problem! If anyone else has this issue, remember you have to log out and log back in to have changes to environment variables apply.

closed 2021-07-09 by joshua-matt